wellenvogel/avnav #175: Current and tides

Fietsenwiel, 2021-12-07:
In OpenCPN it is possible to show currents and tide in the map.

Now, AvNav is for me much nicer plotter software (accually great!) so I was wondering if it is also possible to see current and tides in AvNav.

wellenvogel, 2021-12-07:
Unfortunately not yet....
Would be a great idea for some plugin.

bobhy, 2022-06-19:
I'm interesting in making this work, could do some development about it.  

Could you suggest a good example of a plugin to base this on?  

I see that you can create an overlay from KML which can easily be attached to a chart, and I can find KML files from NOAA which show the *locations* of tide stations but not the current observations.  

NOAA does host an online API to generate predictions from geo-referenced stations (for USA) and I can see how to extend the overlay KML handler to cache that data and convert it into renderable tide and current predictions on request.  However, I don't think the overlay request is bounded when the viewer queries the server for overlay data, so it would regenerate the whole region on each chart update -- maybe I should be looking at the chart handler instead?  Guidance as I look into AvNav source code would be appreciated, there is a lot of it!

free-x, 2022-06-20:
small addition for NL and German tides. One of our users has built KML overlay
https://www.segeln-forum.de/thread/43757-raspberry-pi-als-bordcomputer-spielzeug-f%C3%BCr-den-winter/?postID=2382095#post2382095

wellenvogel, 2022-06-20:
Finally it depends  a bit on how you would like to display.
For a simple display as html page after clicking the station ( i.e. feature overlay) I think it should be possible to already let a plugin generate the html on the fly by using some /plugin/... urls in the kml.
The plugin could then query and cache the Api or use the mathematics to compute it (better).
Finally it would be nice if the plugin could just provide the complete overlay... (needs some api extension).
Need to try this first...
Have to try t