wellenvogel/avnav #176: AvNav on latest buster missing rendering tiles, giving many network errors even for local requests

mgrouch, 2022-01-06:
AvNav on latest buster missing rendering tiles, giving many network errors even for local requests.

Somehow it looks broken. I see many failed requests.

I was testing with BBN OS 2022-01-04 (which is latest Buster on raspberry pi)
and NOAA charts in Long Island NY area.

"unable to query overlays: Network Failed"

Network itself is OK. Wired Ethernet with good speed.
224 MBit/sec by OOCLA Speedtest.

Going into server status in AvNav gives "Network request failed"

Thanks

mgrouch, 2022-01-06:
```
self.handle_error()
File "/usr/lib/python3.7/wsgiref/handlers.py", line 368, in handle_error
self.finish_response()
File "/usr/lib/python3.7/wsgiref/handlers.py", line 180, in finish_response
self.write(data)
File "/usr/lib/python3.7/wsgiref/handlers.py", line 274, in write
self.send_headers()
File "/usr/lib/python3.7/wsgiref/handlers.py", line 331, in send_headers
if not self.origin_server or self.client_is_modern():
File "/usr/lib/python3.7/wsgiref/handlers.py", line 344, in client_is_modern
return self.environ['SERVER_PROTOCOL'].upper() != 'HTTP/0.9'
TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/lib/avnav/server/handler/httpserver.py", line 252, in tryExternalMappings
return self.externalHandlers[prefix].handleApiRequest('path', path, requestParam, server=self,handler=handler)
File "/usr/lib/avnav/server/handler/pluginhandler.py", line 626, in handleApiRequest
rt=api.requestHandler(localPath[1][4:],handler,requestparam)
File "/usr/lib/avnav/server/../plugins/mapproxy/plugin.py", line 659, in handleApiRequest
self.mapproxy.handleRequest(url,handler,args)
File "/usr/lib/avnav/server/../plugins/mapproxy/mapproxy_wrapper.py", line 403, in handleRequest
shandler.run(self.mapproxy)
File "/usr/lib/python3.7/wsgiref/handlers.py", line 144, in run
self.close()
File "/usr/lib/python3.7/wsgiref/simple_server.py", line 35, in close
self.status.split(' ',1)[0], self.bytes_sent
AttributeError: 'NoneType' object has no attribute 'split'

2022-01-06 04:58:46,737-750-8669-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/612/768.png": timeout('The read operation timed out')
2022-01-06 04:58:46,916-750-8781-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/612/766.png": timeout('The read operation timed out')
2022-01-06 04:58:46,926-750-8782-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/613/768.png": timeout('The read operation timed out')
2022-01-06 04:58:54,554-750-8797-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/613/766.png": timeout('The read operation timed out')
2022-01-06 05:50:25,206-750-11966-HTTPHandler-INFO-added channel connection for 0
2022-01-06 05:50:25,208-750-11966-HTTPHandler-INFO-STATUS[0] NMEA, enabled 1 connections
2022-01-06 05:51:28,247-750-11958-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1225/1533.png": timeout('The read operation timed out')
2022-01-06 05:51:28,321-750-11959-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1226/1533.png": timeout('The read operation timed out')
2022-01-06 05:51:31,315-750-11995-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1225/1531.png": timeout('The read operation timed out')
2022-01-06 05:51:31,327-750-11997-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1226/1531.png": timeout('The read operation timed out')
2022-01-06 05:52:10,070-750-12020-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1227/1531.png": timeout('The read operation timed out')
2022-01-06 05:52:12,335-750-12023-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/612/766.png": timeout('The read operation timed out')
2022-01-06 05:52:32,584-750-12064-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/10/305/382.png": timeout('The read operation timed out')
2022-01-06 05:52:35,335-750-12071-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/606/767.png": timeout('The read operation timed out')
2022-01-06 05:52:36,244-750-12078-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/11/607/767.png": timeout('The read operation timed out')
2022-01-06 05:52:36,466-750-12082-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1211/1536.png": timeout('The read operation timed out')
2022-01-06 05:53:10,354-750-12100-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1210/1535.png": timeout('The read operation timed out')
2022-01-06 05:53:12,482-750-12102-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1211/1535.png": timeout('The read operation timed out')
2022-01-06 05:53:22,891-750-11966-HTTPHandler-INFO-STATUS[0] RUNNING, enabled 0 connections
2022-01-06 05:53:35,483-750-12119-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/12/1211/1537.png": timeout('The read operation timed out')
[... cut ...]
2022-01-06 06:10:00,394-750-13181-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/14/4837/6149.png": timeout('The read operation timed out')
2022-01-06 06:10:27,196-750-13202-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/14/4837/6150.png": timeout('The read operation timed out')
2022-01-06 06:10:31,006-750-13208-HTTPHandler-INFO-MapProxy: could not retrieve tile: Internal HTTP error "https://tileservice.charts.noaa.gov/tiles/50000_1/13/2419/3074.png": timeout('The read operation timed out')
```

mgrouch, 2022-01-06:
Well I picked one of tile requests and issued it from chromium

https://tileservice.charts.noaa.gov/tiles/50000_1/14/4838/6149.png

It took very long time to return. But it returned.
Might be AvNav should have longer timeout. At least if people use it underway
they would get it eventually instead of failing.

mgrouch, 2022-01-06:
Interestingly TukTuk chartplotter works just fine in same conditions.

free-x, 2022-01-06:
```
$ time curl https://tileservice.charts.noaa.gov/tiles/50000_1/14/4838/6149.png
curl: (56) OpenSSL SSL_read: Connection reset by peer, errno 104

real	2m2,412s
user	0m0,063s
sys	0m0,013s
```
I think NOAA should be the first to fix this.

mgrouch, 2022-01-06:
> ```
> $ time curl https://tileservice.charts.noaa.gov/tiles/50000_1/14/4838/6149.png
> curl: (56) OpenSSL SSL_read: Connection reset by peer, errno 104
> 
> real	2m2,412s
> user	0m0,063s
> sys	0m0,013s
> ```
> 
> I think NOAA should be the first to fix this.

Unless they introduced some kind of throttling and started denying some requests from clients they think 
issue too many too often. Navionics does something like that, so that's a possibility.

free-x, 2022-01-06:
Thx for dig in right way
You must set only Cookie and it works like a sharm
```
time curl -vvv -A "Mozilla/5.0 (Windows NT 6.1; Win64; x64; rv:59.0) Gecko/20100101 Firefox/59.0" -H "Cookie: tileset-page=%2350000_1-locator; tileservice-disclaimer-accepted=true" -k https://tileservice.charts.noaa.gov/tiles/50000_1/14/4838/6149.png --output 6149.png
....
real	0m0,826s
user	0m0,082s
sys	0m0,026s
```

wellenvogel, 2022-01-07:
Funny enough - tried it right now with the URL you had - 
`time curl -o /dev/null https://tileservice.charts.noaa.gov/tiles/50000_1/14/4838/6149.png
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3597  100  3597    0     0   5525      0 --:--:-- --:--:-- --:--:--  5525

real	0m0,682s
user	0m0,018s
sys	0m0,011s
`
Hmm - strange - no issues (tested on my desktop - ubuntu18.04).
And in AvNav - it's a bit slow - but no errors...
Anyway the issue is in the wrong repo - it should go to 
https://github.com/wellenvogel/avnav-mapproxy-plugin.
Not sure if it would make sense to add some header options to the mapproxy config for noaa...

wellenvogel, 2022-01-07:
We should folllow at the mapproxy-plugin:
https://github.com/wellenvogel/avnav-mapproxy-plugin/issues/8

closed 2022-01-07 by wellenvogel