wellenvogel/avnav #177: Add a parameter to the signalk plugin to choose between http and https

mhomeijer69, 2022-01-08:
In the signalk plugin, http is hardcoded. Could you add an option to the plugin to choose http or https? When I change the plugin in my avnav setup to connect to a secured signalk instance, it's working.

free-x, 2022-01-08:
@mhomeijer69 
could you send us output of
curl https://<server>:<port>/signalk
if your server has self-signed certificate, you should use option "-k" for curl

mhomeijer69, 2022-01-09:
The output is: {"endpoints":{"v1":{"version":"1.40.0","signalk-http":"https://signalk.yachtsymbiant.net:3443/signalk/v1/api/","signalk-ws":"wss://signalk.y
achtsymbiant.net:3443/signalk/v1/stream","signalk-tcp":"tcp://signalk.yachtsymbiant.net:8375"}},"server":{"id":"signalk-server-node","versio
n":"1.40.0"}}

wellenvogel, 2022-01-09:
OK, will have a look at this.

mhomeijer69, 2022-02-20:
I see signalk data coming in in AvNav, but when I change the layout of AvNav, I can't select any signalk data. Could this be related to https or could something else be wrong here?

wellenvogel, 2022-02-20:
Not sure how your setup looks like.
How do you see data coming in to AvNav?

mhomeijer69, 2022-02-20:
By adding debug statements in the signalk plugin.py: 2022-02-20 20:27:35,778-16338-16387-signalk-websocket-INFO-Add data gps.signalk.navigation.speedOverGround, 4.1670010556402675

wellenvogel, 2022-02-21:
Could you just send the full debug output - and maybe some trace from the web access?

mhomeijer69, 2022-02-24:
Okay, forget that last one. I was confused by the fact that some widgets (like COG) only use the internal value and not the signal k value. Sorry for the confusion.

Conclusion so far that if you put https in the signalk url, get the signalk certificate validated (probably needs a library) or disable the verification (_create_unverified_context), the signalk plugin will run against an ssl signalk server. Next goal is to get avnav behind https.