wellenvogel/avnav #179: Improve developer documentation

emilecantin, 2022-01-12:
I'm currently trying to get avnav up & running on my main computer for development purposes (I'm a React dev and I've spotted a few usability issues I'd like to submit a PR for), but I can't really find out how.

I'd like to mention I've got the "release" version up & running on my Raspberry Pi using OpenPlotter, this is really about a dev environment.

Here's what I've tried:

- I've read the top-level README
- I've followed the links to try and find how to get a development setup going (no success)
- Read a few of the READMEs in the lower-level folders
- I noticed some gradle-related files at the top level; I assume that's the build system used on this project
- I ran `./gradlew` => complained about missing Java (obviously)
- I installed the latest Oracle JDK, re-ran `./gradlew` => cryptic error, apparently it's the wrong Java version
- Uninstalled that JDK and installed an older OpenJDK version (plus some PATH shenanigans)
- re-ran `./gradlew` => build successful, but not much seems to have happened
- Inspected the output of `./gradlew tasks`, tried a few of them
- `./gradlew beta` seems promising (built a bunch of node-related stuff)
- Went into the `viewer` folder and tried the various NPM scripts, all of them only build files and none actually start a dev server (that's usually the convention on webpack projects with the `watch` script)
- Went into the `server` folder and tried to run the thing. Unclear what is the main script, ran the one that's executable but the error is probably about a missing dependency (as I'd expect, but there's no requirements.txt so I have no idea how to install them; I'm not a python guy)
- Tried opening the built HTML files from the `viewer` folder, no luck again.

As you can see, I'm usually a resourceful guy, but I'm stumped here. It's probably something simple I've missed, but there's no mention of it in the "standard" places. Can anyone provide more clarity on how to setup a development environment?

wellenvogel, 2022-01-12:
Completely agree...
But: you are the first one asking for this kind of things...
Small hints:

- Tested and running with java 8, gradlew beta is the right thing. This will build all the packages and the android app (beta version).
  If you only want to play with the viewer (js part) - just `npm install && npm run watch` in the viewer dir. Just updated to the latest nodejs (16) - still not tested if it will safely run with older versions.
- to start a server you can use the "avnav" script - assuming you changed into the base dir:
 `linux/avnav -a . -v viewer/build/debug/ -c ~/AvNavCharts/out`
  The -v should either point to debug or release, the -c would this way give you the charts you potentially created with the converter. You can omit it if you don't need those. -a is just for dev purposes and points to the software base.
  Default url is http://localhost:8081.
- I typically run `npm run watch` in the viewer dir and the command above to run the server. Dev server is not set up - would anyway need to start the real server somehow - or implement a lot of mocks for the api calls
- for the requirements:
  Just have a look into the [build.gradle](https://github.com/wellenvogel/avnav/blob/f5da6b7e1697facb922e36563173733c693c6498/build.gradle#L166) - there you find the requirements needed to install the package - just install them on your dev system.

Hope this helps.

emilecantin, 2022-01-12:
No worries, a lot of my own personal (and some professional) projects have minimal / non-existent dev documentation. I'll probably submit a PR on the README with what worked for me once I get this working.

It seemed to work with OpenJDK 11 (that's the one I could easily install with Homebrew), but it just skipped the Android stuff (fine by me, I'm not looking to work on that).

I _am_ mostly interested in the viewer, and I got it to build, but I can't really run it without a server. The usual setup I've seen in most of my front-end projects is using the dev-server, but proxying API calls to a backend server. This allows more flexibility in terms of _where_ you run the backend server (just change an env var to point it to a production / staging instance, for example), and would've let me use the version installed on my Raspberry Pi as a backend.

I'm running this on a mac, so I'll admit I haven't looked at the linux directory. Looks like the mac equivalent script doesn't do as much "stuff"; I assume it means it's not as maintained as the linux version, so probably out-of-date.

In any case, the linux script fails with a bunch of readlink "illegal option" errors (probably because of the BSD vs GNU differences in a lot of the base commands).

The osx script fails with the same python errors as when I ran the python script directly (makes sense, it's only starting that script with a few arguments).

I don't want to install too much python stuff I'll never reuse on my system (I'm really not a python guy), so my next steps will probably be to try and run this in a Docker. It'll let me use the more-maintained Linux version, and we'll hopefully end up with an easier setup for the next guy.

This has gotten pretty long (apologies!), so to sum up my next steps:

1. I'll set up the dev-server in the "watch" Webpack scripts, and configure a proxy for API calls whose target can be overriden with an environment variable
    - This will let me run a front-end dev environment against my "production" backend on my RasPi
2. I'll try to get the Linux version of the server up & running in a Docker.

I'll submit separate PRs for both.

emilecantin, 2022-01-12:
Thanks for confirming my suspicions.

You're right about websockets, they do need special consideration but I've made them work on a previous project; it's not too much of an issue (might need to use a ~10-line "custom" proxy handler).

I think I prefer Docker, mainly for the fact that I'm more familiar with it than pip and virtualenv.

Good point about disabling stuff in the config, it'll probably make it easier to get going. The python error I'm getting is actually related to udev, so if that can be disabled I might be able to go a bit further.

Anyway, I'll work on this tonight and I'll report here if I still have questions.

Thanks for the help!

AWeerWolf, 2022-05-06:
Hi,
I am currently in the same process on windows.
No success yet. 
I can start the app (npm run watch) but cannot access the View page (localhost:8081) probably because no server started.
I did try the gralew.bat but no success. (exception in UploadHtaccess.groovy)
Any idea? 
Should I make a VirtualMachine with Linux and go it or can I stay in Windows?

wellenvogel, 2022-05-07:
Currently you need to run the server in any case.
On windows that would mean to directly start the python part ( on linux you have the linux/avnav script).
I will try to collect the necessary command line options - basically you need to set the path to the compiled js...
If you want to build packages you can run
gradlew.bat beta
But this will not help you for the dev setup.

wellenvogel, 2022-05-07:
OK, just did some testing to give you the shortest possible start command...

1. create a data directory, copy the attached file as avnav_server.xml there 
[avnav_server.txt](https://github.com/wellenvogel/avnav/files/8645813/avnav_server.txt)
2. set DATADIR=your\created\dir
3. run the npm watch in the viewer dir
4. cd into the avnav directory
5. python.exe  server\avnav_server.py -l INFO -w %DATADIR% -u viewer=viewer\build\debug %DATADIR%\avnav_server.xml

Most probably you must prefix "python.exe" with the path to your python3.
Now you should be able to open the web page at localhost:8081.

AWeerWolf, 2022-05-08:
almost good!
When starting the server with python got the following error :
ModuleNotFoundError: No module named 'serial'

AWeerWolf, 2022-05-08:
H:\Documents\BizzKeys\Devel2022\AVNav\avnav>echo %DATADIR%
H:\Documents\BizzKeys\Devel2022\AVNav\avnav\data

H:\Documents\BizzKeys\Devel2022\AVNav\avnav>python server\avnav_server.py -l INFO -w %DATADIR% -u viewer=viewer\build\debug %DATADIR%\avnav_server.xml
Traceback (most recent call last):
  File "H:\Documents\BizzKeys\Devel2022\AVNav\avnav\server\avnav_server.py", line 36, in <module>
    from handler.baseconfig import AVNBaseConfig
  File "H:\Documents\BizzKeys\Devel2022\AVNav\avnav\server\handler\__init__.py", line 16, in <module>
    importlib.import_module("."+regexp_result.groups()[0],__package__)
  File "D:\Python310\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "H:\Documents\BizzKeys\Devel2022\AVNav\avnav\server\handler\feeder.py", line 29, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

wellenvogel, 2022-05-08:
Pip install serial...

AWeerWolf, 2022-05-08:
Ooopsss,, yes did it.
I got unable to parse config file=> no fallback found

2022-05-08 17:53:21,940-31300-10440-MainThread-ERROR-unable to parse config file H:\Documents\BizzKeys\Devel2022\AVNav\avnav\data\avnav_server.xml, no fallback found
AVNAV-ERROR: unable to parse config file H:\Documents\BizzKeys\Devel2022\AVNav\avnav\data\avnav_server.xml, no fallback found

AWeerWolf, 2022-05-08:
All working fine!!! 
had a error  while copying the xml file.
thanks a lot.

wellenvogel, 2022-05-08:
Happy coding...

wellenvogel, 2024-02-19:
No further activity - so closing this.

closed 2024-02-19 by wellenvogel