wellenvogel/avnav #181: oeRNC chart support

emilecantin, 2022-01-17:
While looking at the o-charts shop for charts of the areas I want to cruise (Bahamas), I noticed some of them are only available in oeRNC (raster) format. Looks like these use a similar, but separate plugin from the "normal" oeSENC charts in OpenCPN.

I assume it'd be possible to build a plugin for AvNav similar to the one we have for oeSENC charts? If anything, I suspect it'd be even simpler as we wouldn't have to pre-render the charts.

wellenvogel, 2022-01-17:
The o-charts plugin will be able to handle this (with some limitation in handling chart boundaries) once the o-charts shop is ready in the transition to their new format.
This was already planned for a long time (several months) - so we have to wait.
Please follow up at the [o-charts plugin](https://github.com/wellenvogel/avnav-ocharts-provider).

emilecantin, 2022-01-17:
Interesting, I wasn't aware that o-charts was planning on a format change. Is this discussed somewhere on the Internet? Their website makes absolutely no mention of it.

free-x, 2022-01-17:
https://github.com/bdbcat/o-charts_pi
`This is a plugin for opencpn [1] providing support for encrypted charts available from o-charts.org [2]. The plugin supports purchase, downloading and rendering of these charts.

o-charts can handle charts which have been handled by the oesenc and oernc plugins and thus obsoletes these plugins.`

wellenvogel, 2022-01-17:
Ok, I will close this here as there are no actions on AvNav itself.

closed 2022-01-17 by wellenvogel