wellenvogel/avnav #183: v20220212: Boat direction mode HDT not working when more-nmea plugin is used

W-Geronius, 2022-02-13:
In my setup magnetic heading is propagated from pypilot (moitessier hat) via SignalK to AvNav. The values correctly arrive in gps.headingMag. The avnav-more-nmea-plugin is used to calculate variation and HDGm and HDGt.

relevant output of http://10.10.10.1:8080/viewer/avnav_navi.php?request=gps

> {
> "headingMag": 36.3,
> "MagVar": 2.7185108949261823,
> "HDGm": 36.3,
> "HDGt": 39.01851089492618,
> }
> 

The navlayer.js code shows that gps.headingTrue is used to determine boat direction mode HDT. This value however remains undefined and direction mode defaults to COG.

I tried a dirty fix in the plugin.py, outputting true heading to gps.headingTrue instead of HDGt, with no success. Is there any architectural reason, why the plugin isn't populating AvNav's standard variables (headingMag and headingTrue)?

wellenvogel, 2022-02-13:
Yes, plugins cannot overwrite AvNav's internal data until explicitly enabled to do this.
Maybe we could add an option to the plugin to send back it's created NMEA data to AvNav's decoder.
Would simple mean to include omitDecode=False in the calls to ...api.addNMEA(...).
The default for this flag is True.
This way the HDG/HDT created by the plugin would be decoded by AvNav itself.

W-Geronius, 2022-02-13:
as an alternative to modifying the plugin (I don't know if author is still active), it might be an option to use gps.magVar and gps.headingMag to set gps.headingTrue internally within AvNav unless set otherwise

W-Geronius, 2022-03-14:
fixed with merge of PR mentioned above

closed 2022-03-14 by W-Geronius