wellenvogel/avnav #184: v20220212: storekeys, default, etc. get lost when updating user widgets

W-Geronius, 2022-02-17:
Somewhere on its way from last public in summer 2021 to version daily 20220212 an error was introduced in the layout editor.
Whith various widgets some parameters (as storekeys, default, and possibly others) are not written back to the layout json file when updated. Inserting new widgets works ok.

Example: 
select default user 'example' layout and update to 'example_updated #3' Enter gpspage 3, update all widgets with no changes and save layout. See results in screenshot:

[image]
[json.zip](https://github.com/wellenvogel/avnav/files/8087356/json.zip)

wellenvogel, 2022-02-17:
I made some changes there - thanks for reporting this!
Will have a look...

W-Geronius, 2022-02-18:
checked with [daily-20220217](https://github.com/wellenvogel/avnav/releases/tag/daily-20220217):
most of problems have disappeared. The problem however persists with history widgets:

        {
          "caption": "AWS",
          "unit": "knots",
          "fieldName": "gps.signalk.environment.wind.speedApparent",
          "color": "#0e31f2",
          "fieldFormatter": "knots",
          "hours": "72",
          "yMin": "20",
          "yMax": "25",
          "showLines": true,
          "name": "HistoryWidget",
          "weight": 1
        },
        {
          "caption": "AWS",
          "unit": "knots",
          "color": "#0e31f2",
          "fieldFormatter": "knots",
          "hours": "72",
          "yMin": "20",
          "yMax": "26",
          "showLines": true,
          "name": "HistoryWidget",
          "weight": 1
        },

reproduce: 
take first widget, add it 'after' unchanged. Works ok.
edit second widget, set yMax to '26' -> fieldName is lacking after 'update'

I haven't checked other 'special' widgets so far

W-Geronius, 2022-02-19:
I have no indication it's caused by default logics. Creating a new history widget (either new or by copying) works perfectly. When updating that widget (with all fields correctly set on screen), entry "fieldName" is lost in the JSON file

W-Geronius, 2022-02-20:
updated to daily 20220219 - to improve test coverage, I created a (otherwise totally useless) 'WidgetSetupTest' layout holding all available widgets spreaded over gpspages 1-5:
[WidgetSetupTest.json.txt](https://github.com/wellenvogel/avnav/files/8104098/WidgetSetupTest.json.txt)

In edit mode every single widget was opened and updated (with no changes yet), the resulting layout  being saved as 
[WidgetSetupTest_updated.json.txt](https://github.com/wellenvogel/avnav/files/8104118/WidgetSetupTest_updated.json.txt)

Comparing original and updated version expose many differences (omissions)

wellenvogel, 2022-02-20:
Anything not working as expected?
A lot of changes are completely ok...
I chnage the handling a lot to prevent saving of data that is equal to the widget defaults.
This allows for widgets to change their defaults ( typically you would expect this to become  active for all widgets). With the old behavior it could become a bit unpredictable.

W-Geronius, 2022-02-21:
Sorry, I can't yet tell which ones arent working, but I will check soon as far as I can. I have no NMEA0183 testdata generator in place so far.

I was surprised about the file content changes, because I had created WidgetSetupTest layout from scratch with a fairly current version (20220212 or so). I used system default, deleted ALL widgets contained therein and inserted one by one, following the 'new widget' dropdown list. Sorry for the bother, if there is a misconception on my side

W-Geronius, 2022-02-23:
I have completed widget update test using v20220219. No further update errors were noted - all widgets remain operational.

closed 2022-02-23 by W-Geronius