wellenvogel/avnav #188: One master device configuring n remote devices

pjsf1000, 2022-02-23:
I would like to have a master AVNav device on the boat and then use that to control what the other devices display.  The remote function gets me part of the way there as long as the remote devices display exactly whats on the master device.  Ideally I would have each device on a different remote channel, use the master device to control what is displayed on that remote channel and then disconnect the master device from controlling that channel, so the two devices can display different screens.  If/When I currently try to disconnect the remote channel the remote device follows my control to the disconnect screen rather than the remote device staying on the screen i want it too.  

IMHO: A simple implementation would enable the user to disconnect the remote connection on the master device from any screen, thus leaving the remote device on the configured UI.  A more elaborate implementation would allow the master device to use the 2nd split screen to swipe through each of the remote devices. 

BTW: AVNav is stunning solution and I can't wait to use it this coming season on the water,  thanks for creating it!

wellenvogel, 2022-02-23:
Very valid point.
The only problem is the number of buttons at the pages.
But most probably I will add a toggle to at least main page, navigation page and dashboards.
Most probably with a small dialog for attach/detach/channel/rw.
This will not be forwarded via remote - so the other display will remain at it's current page.
The remote handling for the split screen still needs some thoughts at all - still not sure what the best handling approach will be.
At least you would most probably like to be able to switch the remote window somehow - but things are even getting more interesting if you have split screen on one display but not on the other...

wellenvogel, 2022-04-21:
release 20220421

pjsf1000, 2022-04-26:
Hi, I've just tried this out and it solves the problem really well.   From a central nav station I can now control the screen of multiple devices around the boat.. Many thanks for implementing this feature!