wellenvogel/avnav #200: suggestion: layout/navigation: boat direction

W-Geronius, 2022-03-14:
after the great update in 2021 and referencing #88,  please consider a further enhancement:
presently users are given the choice between COG and HDG view. Imho, HDM view should only be set in areas with neglectible variance.

The grafic representation of the boat's movement (COG) and orientation (HDG) could be refined by accounting for the data available, using this ruleset:

[image]

W-Geronius, 2022-03-14:
I have also setup a matrix to include STW, but that in fact doesn't really add more information

W-Geronius, 2022-04-04:
next iteration after discussion
[image]

wellenvogel, 2022-04-05:
Hmm, at least a bit different then what we discussed.
I thought we discussed the following options:
(1) always COG
(2) use HDT (fallback to COG)
(3) use HDM (fallfack to COG)
This is what we have right now.
We will just add a different boat image if using HDT or HDM.
We add a new preference (maybe better 2...):
(4) detect steady
(5) steady SOG limit
If detect steady is activated we will switch the boat to a circle if the steady condition is detected (i.e. SOG unavailable or < limit).
Maybe it's easier to separate the table into separate ones depending on the conditions.
[image]
We could still decide what should happen if we have SOG but no COG (and no HDX) and steady dection is on.
Drawing a circle in this case could be slightly confusing as there is movement. But as we discussed this is very unlikely any way - so maybe we just stick at "D" under this condition.

wellenvogel, 2022-04-05:
After some further discussion:
We keep the selections (1)...(3) as they are.
We add some "steady" detection that will give us a hint that COG is not really valid (no SOG or SOG below limit - with some averaging).
We will have 3 boat icons:
(a) D - arrow as it is now, is showing COG (with course vector)
(b) O - new boat icon showing the boat direction (HDT or HDM) with course vector, optional COG dotted vector as now
(3) U - circle icon for steady boat, no course vector

The "steady detection" will be switched on and of via settings.
There will be the option to adapt all the icons as you can do for the boat icon right now already.

[image]

W-Geronius, 2022-04-05:
perfect, thank you!

closed 2022-04-05 by W-Geronius