wellenvogel/avnav #206: remote control anchor alarm activation

mat214, 2022-04-01:
Hello,
Is it possible to trigger anchor monitoring from a script on the position of the boat?

The idea is to trigger this monitoring by pressing a bluetooth remote control like this one: https://www.amazon.fr/gp/product/B07CXFX2HM/ref=crt_ewc_title_oth_1?ie=UTF8&psc=1&smid=A3KRSE5WB004NH

The method exists for signal k: [https://github.com/meri-imperiumi/signalk-bluetooth-anchor-button](https://github.com/meri-imperiumi/signalk-bluetooth-anchor-button)

Thus the crew member who is at anchor presses the button at the right time.

Thanks for your help.

wellenvogel, 2022-04-02:
Very nice idea...
There is for sure a couple of ways to achieve this.
Maybe as a first shot I will add some [key commands](https://www.wellenvogel.net/software/avnav/docs/hints/keyboard.html?lang=en) to start and stop anchor watch.
Then you can assign keys to this command (I will already add some defaults) and use the UDP receive function of the [remote channel control](https://www.wellenvogel.net/software/avnav/docs/hints/RemoteControl.html?lang=en#h1:RemoteControl) to trigger.
Drawback of this approach is that you still need a running(!) browser with AvNav open being connected to remote channel 0.
But it's fairly easy to use it - and you could also use keys on a device to trigger this.
A next/different approach would be to extend the [plugin api](https://www.wellenvogel.net/software/avnav/docs/hints/extensions.html?lang=en#h2:Plugins) to allow to set the anchor watch.
This would give you the option to write some lines of python code to that directly would be controlled by AvNav and call the API on whatever event you like.
Advantage: No running browser needed as it runs totally on the server.
I guess finally I will go for both.
Anchor watch and MOB should be at the plugin api anyway together with some alarm handling stuff

wellenvogel, 2022-04-08:
With the current approach you can write a script that send a keycode command via UDP (e.g. using the nc command).
It's still not perfect as I wrote - you just need a running browser with AvNav that is connected to remote channel 0.
Assuming you have everything set  up as default:
(1) let your browser listen on remote channel 0 (the remote button will allow to set this - or in settings)
(2) run on the pi:
```
echo 'K i' | nc -w0 -u localhost 34668
```
This sends the key "i" on AvNav's remote channel 0 - by default configured to switch on anchor watch at the current position.

Until now you cannot cange the radius afterwards.
But with some simple steps it should be possible to do what you want:
(1) on the navpage click on the lower left displays, that will bring up the additonal waypoint buttons. With anchor watch being active this is only anchor and center
(2)click the center button (below anchor) - will center the map at the current anchor position
(3) switch of anchor watch
(4) switch on anchor watch - (maybe slightly moving the center before) - select the radius and whatever you need and click "center" - this will activate the anchor watch again at the current map center.

Not ideal until now - but was a fast solution.
When a new plugin api will be available it's just about some lines of python code to do this. Maybe I will go for some extensions in the features of the UDP interface of the remote channels to also do some things on the server...
Finally you may also want to get some response if the action was successful.

You should not try to use a post request as the one you saw - this is some internal API in AvNav and can easily change with some update.

mat214, 2022-04-08:
Thanks again for the response. By doing tests and plugging an earphone on my rpi I don't have the sound of the alarms from the server.
However the output works with: mpg123 /lib/avnav/sounds/anchorAlarm.mp3

Is it necessary to have a particular configuration to have the sound output of the alarms on the rpi?

wellenvogel, 2022-04-08:
Which installation do you use?
Is it our headless image?
And most probably you  are running the latest daily?

mat214, 2022-04-08:
J'ai fait l'installation de l'image sans tête comme expliqué sur votre site : https://www.free-x.de/raspbian/bullseye/

wellenvogel, 2022-04-08:
Ok, I see.
I guess I already corrected this. Most probably a new daily will be there today in the evening - this should make the sound working again.

wellenvogel, 2022-04-21:
1st step implemented with release 20220421. Use the remote control feature...
Next step will be a plugin API.

closed 2025-07-23 by wellenvogel