wellenvogel/avnav #209: Directory List for Charts

AWeerWolf, 2022-04-14:
It would be nice if we can specify list of directories for the charts. 
This could be useful to put the chart on a separate partition and also share them with OpenCPN.

wellenvogel, 2022-04-14:
Do you talk about the charts that you directly use in AvNav (for o-charts you can already easily share them with OpenCPN)?
Currently you can set the chart directory at AVNHttpServer...
Something like
```
<AVNHttpServer>
....
<Directory path="/my/special/folder/for/charts" urlpath="maps"/>
</AVNHttpServer>
```
Just refer to the existing avnav_server.xml for the current values.
You can use the update plugin to edit the config file.
But multiple chart dirs are still not possible....

AWeerWolf, 2022-05-04:
Hello,

I did what you suggested : Specify a path for the Maps (on a different partition)
There is a mbtiles file in the specified directory.
I did restart AVNav server and refresh the web page, but I do not see the map.
Any idea ???

By the way, it seems that the ChartHandler is still pointing to the old chart directory.

Regards,
[image]
[image]
[image]
[image]

wellenvogel, 2022-05-05:
Ok, I have to cross check.
Basically this is no "official" way of doing it - so not sure if it still works.

wellenvogel, 2022-05-12:
Currently the chart dir is determined by the -c paramter when AvNav is started.
You could go for a systemd service override to change this.
I will have a look if this parameter can be removed in the default set up - this should make the setting from avnav_server.xml eorking again.

wellenvogel, 2022-05-13:
What you can do for now:
Create a (systemd) config override with the following commands
```
sudo mkdir /etc/systemd/system/avnav.service.d
sudo nano /etc/systemd/system/avnav.service.d/zzcharts.conf
sudo systemctl daemon-reload
sudo systemctl restart avnav
```
When the editor (nano) is open, just insert:
```
[Service]
ExecStart=
ExecStart=/usr/bin/avnav -q -c /home/pi/avnav/charts2 -b /home/pi/avnav/data -t /usr/lib/avnav/raspberry/avnav_server.xml
```
Use the path to your chart directory for the "-c" parameter.
Import to have both of the ExecStart= lines (the first one with empty parameter)!
The name of the file is important to be the last one considered (zzcharts.conf in my example).
This will continue to work even if I change the behavior as described later on...

AWeerWolf, 2022-05-15:
Works great.
Just one modification, for the "-t" parameter I used /home/pi/avnav/data/avnav_server.xml

wellenvogel, 2022-05-15:
Bad idea...
-t is the template if the config file at the destination location does not exist.
But maybe i missed the last part of the start command - this is the name of the config file ;-(

AWeerWolf, 2022-05-15:
In /usr/lib/avnav/raspberry is empty except there is only one file : sound.sh

wellenvogel, 2022-05-16:
Ok, this is for the normal raspberry installation.
If it works for you leave it as you have it.

AWeerWolf, 2022-06-05:
Just an additional info :
As the partition containing the charts is not completely mounted when the system try to start the AVNav service, this one fail to auto start.
To solve this I did add the following lines to the "conf" file :
Restart=on-failure
RestartSec=60
StartLimitBurst=3
StartLimitInterval=400

With these lines, the system will try 3 times to start the service and wait 60 sec between each start.

wellenvogel, 2022-06-05:
Would be better to use correct dependencies.
AvNav will happily start with an empty chart dir...

AWeerWolf, 2022-10-11:
Hello,

I did what you suggested : Specify a path for the Maps (on a different 
partition)

There is a mbtiles file in the specified directory:

I did restart AVNav server and refresh the web page, but I do not see 
the map:

Any idea ???

By the way, it seems that the ChartHandler is still pointing to the old 
chart directory :

Regards,