wellenvogel/avnav #214: Improvement: transparent panels in Layout editor

PietKnot, 2022-05-10:
there is some inconsistensy in the handling of the top/left/bottem panels in the Layout editor / Map presentation.

The perfect implementation is the left panel where unused parts are transparent an the map is shown. Widgets just take the space they need.

Most unfortunately, the bottm left / right do not have this transparency and widgets are often enlarged to fill the available space / ot the panel is left white. Would be nice if the bottom parts would behave like the left pasrt (filling bootm-right from the right, and bottom left from the left). See example screen display
[image]

wellenvogel, 2022-05-11:
I guess most of the things can already be achieved with some user defined css.
First set "float map behind buttons" in settings.
Afterwards add the following to your user.css (and reload after saving).
```
.mapPage .bottomSection .widgetContainer {
    background-color: #ffffff00;
}
.bottomLeft .widget {
   flex-grow: 0;
}
.bottomRight .widget {
  flex-grow: 0;
}
```
Will still fill from the middle.
With the transparent background it should also be fine to just enable one panel.
With some further CSS tuning you could also make one panel fit the complete bottom part - or position it wherever you want.

PietKnot, 2022-05-12:
will try and give feedback

PietKnot, 2022-05-12:
Great solution:  did it and works (a bit try-and-error for the top but oke. Result:

[image]

Just noted that the Widgets in the right-panel are now shifted to the left-panel but i don't care. Will just disable the bottom section `completely;` just keep left and top.  My strong recommendation is to implement this in the default: much better.

`/*
  css file to adapt the avnav look and feel to your needs
  use your browsers development tools to find out about the css classes of every element in the UI
*/
.mapPage .widgetContainer.top {
    background-color: #ffffff00;
}
.mapPage .bottomSection .widgetContainer {
    background-color: #ffffff00;
}
.top .widget {
   flex-grow: 0;
}
.bottomLeft .widget {
   flex-grow: 0;
}
.bottomRight .widget {
  flex-grow: 0;
}
`

wellenvogel, 2022-05-12:
User.css on android works the same way.
Reloading is a bit more tricky as you have to restart AvNav.
Would recommend to use a browser to tune the user.css ( I always use 2 tabs - one with the editor open and one to test and reload).
To find css settings I recommend using the developer tools from chrome/firefox/edge to find the things to change.
For the defaults:
People have different preferences on that.
The current ones worked for me for the main focus devices:
Tablets from 7 to 10 inch in landscape mode.
But I understand that others feel that things should look different.
Btw:
If you prefix your classes with the layout you can limit the effect to particular layouts - see http://wellenvogel.de/software/avnav/docs/hints/usercss.html?lang=en

PietKnot, 2022-05-12:
Thanks for the tips . . . . did read that under Android, access to user.css is via Files/Download and Upload (after deletion) ; that worked for me. I do realize that i still have to do some learning (e.g. java / css) to fully configure AvNav as i like.  Issue closed for me!

closed 2022-05-12 by PietKnot

wellenvogel, 2022-05-13:
No need to download/delete/upload.
You can directly use the integrated editor...