wellenvogel/avnav #216: Dashboard COG / BRG extension

PietKnot, 2022-05-12:
on dashboard "3" there are three displays of COG: number, linear, compass; quite a choice. I would like to suggest to improve this by presenting the difference between BRG end COG (like XTE) on the linear scale, or possibly by a green line (like the Red line) on the compass, It gives a direct visual indicating in which direction to manual steer (SB <-> BB) to reach the next waypoint; ofter easier than XTE! 

Another -better?- choice could be in dashboard "1" instead of the GPS time (is already on dashboard  "2")

Since many years in use on very old WindowsCE and Android devices running OziExplorer with fully user configurable map/dashboard display (to be honest, easier to design than AvNav with OZI screen designer for Android / WindowsCE), but  . . . . maps are an issue here.

Example by photo (screenschot on WinCE-a bit cumbersome):
[image]
[image]

wellenvogel, 2022-05-12:
I was also using ozi before I started with AvNav....
For the widgets:
You can easily do most of those things by your own with a couple of lines of Js code.
The only restriction for the canvas gauges: They only can show one value.
But you could also write an own widget - just drawing everything to a canvas.
Have a look at http://wellenvogel.de/software/avnav/docs/hints/userjs.html?lang=en.
Examples are already there.
I would be happy to integrate some new widgets....

PietKnot, 2022-05-12:
Will have to do some homework / learning java / studying examples; wait-and-see will close this issue accordingly.

B.t.w. discovered today how to use SasPlanet to make up-to-date OZI-charts, thus WinCE  with Ozi is still not as dead as i thought!!

closed 2022-05-12 by PietKnot