wellenvogel/avnav #219: Connection errors

mgrouch, 2022-05-15:
Many errors saying connection is lost (I think related to SignalK connection improvements)

`

2022-05-15 06:21:08,224-751-29771-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:21:08,224-751-29790-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:21:38,274-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:21:38,275-751-29790-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:21:38,276-751-29807-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:21:38,277-751-29790-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:22:08,324-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:22:08,325-751-29807-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:22:08,326-751-29827-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:22:08,326-751-29807-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:22:38,374-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:22:38,374-751-29827-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:22:38,375-751-29827-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:22:38,376-751-29846-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:23:08,418-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:23:08,419-751-29846-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:23:08,420-751-29876-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:23:08,420-751-29846-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:23:38,465-751-29876-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:23:38,466-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:23:38,466-751-29876-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:23:38,467-751-29896-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:24:08,514-751-29896-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:24:08,514-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:24:08,515-751-29896-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:24:08,516-751-29915-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:24:38,557-751-29915-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:24:38,558-751-29915-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:24:38,559-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:24:38,559-751-29987-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:25:08,621-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:25:08,621-751-29987-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:25:08,622-751-30122-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:25:08,622-751-29987-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:25:38,665-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:25:38,666-751-30122-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:25:38,667-751-30143-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:25:38,667-751-30122-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:26:08,701-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:26:08,702-751-30162-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:26:08,702-751-30143-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:26:08,705-751-30143-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:26:38,764-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:26:38,765-751-30162-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:26:38,771-751-30162-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:26:38,766-751-30181-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:27:08,833-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:27:08,834-751-30201-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:27:08,835-751-30181-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:27:08,836-751-30181-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:27:38,877-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:27:38,878-751-30201-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:27:38,879-751-30219-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:27:38,879-751-30201-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:28:08,921-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:28:08,922-751-30219-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:28:08,922-751-30236-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:28:08,923-751-30219-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:28:38,973-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:28:38,974-751-30236-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:28:38,975-751-30254-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:28:38,975-751-30236-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 06:29:09,010-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 06:29:09,010-751-30254-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 06:29:09,011-751-30275-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 06:29:09,011-751-30254-signalk-websocket-None-INFO-websocket receiver None finished
[... 296 lines cut ...]
2022-05-15 07:04:12,567-751-31812-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 07:04:26,675-751-31790-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/8/76/93.png": timed out
2022-05-15 07:04:26,677-751-31791-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/8/78/92.png": timed out
2022-05-15 07:04:26,698-751-31788-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/8/78/93.png": timed out
2022-05-15 07:04:26,702-751-31789-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/8/76/92.png": timed out
2022-05-15 07:04:26,725-751-31793-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/7/38/46.png": timed out
2022-05-15 07:04:42,615-751-31834-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 07:04:42,616-751-31834-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 07:04:42,616-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 07:04:42,617-751-31855-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 07:05:12,656-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 07:05:12,657-751-31855-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 07:05:12,658-751-31876-signalk-websocket-None-INFO-websocket receiver None started
[... cut ...]
2022-05-15 07:05:42,697-751-31894-signalk-websocket-None-INFO-websocket receiver None started
2022-05-15 07:05:42,697-751-31876-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 07:06:12,747-751-31894-signalk-websocket-None-INFO-websocket connection None closed
2022-05-15 07:06:12,748-751-1142-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-15 07:06:12,748-751-31894-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-15 07:06:12,750-751-31913-signalk-websocket-None-INFO-websocket receiver None started

`

wellenvogel, 2022-05-15:
Would be helpful to provide the version on which you encountered this.
Should already be corrected with 20220421.

mgrouch, 2022-05-15:
Versions:

```

user@lysmarine:/var/log $ dpkg -l | grep avnav
ii  avnav                                  20220421                                all          avnav
ii  avnav-history-plugin                   20210525                                all          AvNav plugin to display a history of internal data or values from xdr records
ii  avnav-mapproxy-plugin                  20220320                                all          AvNav plugin to handle online charts with mapproxy
ii  avnav-more-nmea-plugin                 20220405                                all          AvNav plugin to calculate derived data out of internal data or values
ii  avnav-ocharts                          0.9.1.0-1buster1                        armhf        O-charts Plugin for AvNav
ii  avnav-ocharts-plugin                   20220426-raspbian-buster                armhf        avnav-ocharts-plugin
rc  avnav-oesenc                           4.2.17-2                                armhf        Plugin for AvNav
ii  avnav-update-plugin                    20220423                                all          AvNav plugin to handle updates of avnav

```

wellenvogel, 2022-05-15:
Ok, I see. Unfortunately I only applied half of the correction...

mgrouch, 2022-05-16:
Can avnav display charts from NOAA ENC viewer?
https://nauticalcharts.noaa.gov/enconline/enconline.html

They are arcgis pbf format. 

Thanks

wellenvogel, 2022-05-16:
Pbf cannot be displayed.
For the status of the NOAA server:
I don't know...

mgrouch, 2022-05-16:
tried daily build 2022-05-15. Still same errors:

```

2022-05-16 21:37:25,571-32284-32315-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-16 21:37:25,574-32284-1939-signalk-websocket-None-INFO-websocket connection None closed
2022-05-16 21:37:25,574-32284-1962-signalk-websocket-None-INFO-websocket receiver None started
2022-05-16 21:37:25,575-32284-1939-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-16 21:37:55,624-32284-32315-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-16 21:37:55,625-32284-1962-signalk-websocket-None-INFO-websocket connection None closed
2022-05-16 21:37:55,625-32284-1981-signalk-websocket-None-INFO-websocket receiver None started
2022-05-16 21:37:55,626-32284-1962-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-16 21:38:25,666-32284-1981-signalk-websocket-None-INFO-websocket connection None closed
2022-05-16 21:38:25,667-32284-32315-SignalKHandler-INFO-websocket None created at ws://localhost:3000/signalk/v1/stream
2022-05-16 21:38:25,667-32284-1981-signalk-websocket-None-INFO-websocket receiver None finished
2022-05-16 21:38:25,668-32284-2003-signalk-websocket-None-INFO-websocket receiver None started

```

wellenvogel, 2022-05-17:
This daily does not have a correction for the issue.
Next one should contain it.

mgrouch, 2022-05-18:
Version 20220517 still has connection errors. 

```

2022-05-18 14:15:13,069-9864-9877-Config-INFO-STATUS[systemtime] RUNNING, UTC: 2022-05-18T13:15:13.068930
2022-05-18 14:15:13,102-9864-9899-HTTPHandler-INFO-----------------------------------------
2022-05-18 14:15:13,104-9864-9899-HTTPHandler-INFO-Exception happened during processing of request from('127.0.0.1', 46130),
2022-05-18 14:15:13,123-9864-9899-HTTPHandler-INFO-Traceback (most recent call last):

2022-05-18 14:15:13,125-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
self.finish_request(request, client_address)

2022-05-18 14:15:13,126-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
self.RequestHandlerClass(request, client_address, self)

2022-05-18 14:15:13,127-9864-9899-HTTPHandler-INFO- File "/usr/lib/avnav/server/handler/httphandler.py", line 60, in __init__
http.server.SimpleHTTPRequestHandler.__init__(self, request, client_address, server)

2022-05-18 14:15:13,127-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/http/server.py", line 646, in __init__
super().__init__(*args, **kwargs)

2022-05-18 14:15:13,128-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
self.handle()

2022-05-18 14:15:13,128-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/http/server.py", line 428, in handle
self.handle_one_request()

2022-05-18 14:15:13,129-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/http/server.py", line 394, in handle_one_request
self.raw_requestline = self.rfile.readline(65537)

2022-05-18 14:15:13,129-9864-9899-HTTPHandler-INFO- File "/usr/lib/python3.7/socket.py", line 589, in readinto
return self._sock.recv_into(b)

2022-05-18 14:15:13,129-9864-9899-HTTPHandler-INFO-ConnectionResetError: [Errno 104] Connection reset by peer

2022-05-18 14:15:13,130-9864-9899-HTTPHandler-INFO-----------------------------------------
2022-05-18 14:15:13,137-9864-9917-Plugin: system-mapproxy-INFO-MapProxy: using libproj for coordinate transformation
2022-05-18 14:15:13,285-9864-9901-Avahi-INFO-STATUS[avnav-server._http._tcp] NMEA, registered
2022-05-18 14:15:13,773-9864-9917-Plugin: system-mapproxy-INFO-registering external chart provider system-mapproxy
2022-05-18 14:15:13,778-9864-9917-Plugin: system-mapproxy-INFO-started
2022-05-18 14:15:13,782-9864-9917-Plugin: system-mapproxy-INFO-creating mapproxy wsgi app with config /home/user/avnav/data/mapproxy/avnav_include.yaml.normal
2022-05-18 14:15:14,090-9864-9884-allnmea0183-INFO-listening at port address ('0.0.0.0', 34567)
2022-05-18 14:15:14,092-9864-9884-allnmea0183-INFO-STATUS[main] RUNNING, listening at ('0.0.0.0', 34567)
2022-05-18 14:15:14,295-9864-9901-Avahi-INFO-STATUS[avnav-server._nmea-0183._tcp] STARTED, trying to register
2022-05-18 14:15:14,296-9864-9901-Avahi-INFO-trying to register avnav-server for 8080
2022-05-18 14:15:14,379-9864-9917-Plugin: system-mapproxy-INFO-created mapproxy wsgi app
2022-05-18 14:15:14,381-9864-9917-Plugin: system-mapproxy-INFO-STATUS[system-mapproxy] NMEA, mapproxy created with config file /home/user/avnav/data/mapproxy/avnav_include.yaml
2022-05-18 14:15:14,393-9864-9915-Plugin: system-ocharts-INFO-starting provider process
2022-05-18 14:15:14,410-9864-9915-Plugin: system-ocharts-INFO-starting provider with command /bin/sh /usr/lib/avnav/server/../plugins/ocharts/provider.sh -t 5 -d 1 -s 2 -l /home/user/avnav/data/ocharts/provider.log -p 9864 -c 10000 -f 400000 -r 17 -e /usr/lib/avnav/server/../plugins/ocharts/bin -n /usr/lib/avnav/server/../plugins/ocharts/lib/opencpn /usr/lib/avnav/server/../plugins/ocharts/share/opencpn /home/user/avnav/data/ocharts 8082
2022-05-18 14:15:14,430-9864-9937-Thread-14-INFO-PROVIDEROUT: working in /home/user/avnav/data/ocharts

2022-05-18 14:15:15,175-9864-9903-AVNAvahi DBUS loop-INFO-detected new service _nmea-0183._tcp.avnav-server at 2.0
2022-05-18 14:15:15,178-9864-9903-AVNAvahi DBUS loop-INFO-detected new service _nmea-0183._tcp.avnav-server at 3.0
2022-05-18 14:15:15,235-9864-9901-Avahi-INFO-STATUS[avnav-server._nmea-0183._tcp] NMEA, registered
2022-05-18 14:15:17,170-9864-10013-Thread-15-INFO-STATUS[monitor] NMEA, running
2022-05-18 14:15:19,430-9864-9915-Plugin: system-ocharts-INFO-started with port 8082
2022-05-18 14:15:19,431-9864-9915-Plugin: system-ocharts-INFO-registering external chart provider system-ocharts
2022-05-18 14:15:19,432-9864-9915-Plugin: system-ocharts-INFO-STATUS[system-ocharts] STARTED, provider started with pid 9936, connecting at http://localhost:8082/list
2022-05-18 14:15:19,601-9864-9915-Plugin: system-ocharts-INFO-got first provider response
2022-05-18 14:15:19,602-9864-9915-Plugin: system-ocharts-INFO-STATUS[system-ocharts] NMEA, provider (9936) sucessfully connected at http://localhost:8082/list
2022-05-18 14:15:26,653-9864-9893-HTTPHandler-INFO-STATUS[0] RUNNING, enabled 0 connections
2022-05-18 14:15:33,500-9864-10113-HTTPHandler-INFO-added channel connection for 0
2022-05-18 14:15:33,501-9864-10113-HTTPHandler-INFO-STATUS[0] NMEA, enabled 1 connections
2022-05-18 14:15:43,014-9864-9919-Plugin: system-history-INFO-STATUS[system-history] NMEA, writing to /home/user/avnav/data/plugins/history/2022-05-18.avh
2022-05-18 14:16:43,197-9864-10109-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/6/18/24.png": timed out
[... cut ...]
2022-05-18 14:18:43,523-9864-10239-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/5/9/13.png": timed out
2022-05-18 14:18:43,544-9864-10244-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/5/8/13.png": timed out
2022-05-18 14:19:43,796-9864-10310-HTTPHandler-INFO-MapProxy: could not retrieve tile: No response from URL "https://tileservice.charts.noaa.gov/tiles/50000_1/5/10/13.png": timed out

```

wellenvogel, 2022-05-18:
Remaining one are mapproxy errors - they will only occur if you actively use a non working mapproxy source. I will not change the logging of mapproxy at this place.

closed 2022-05-18 by wellenvogel