wellenvogel/avnav #226: Small-landscape-layout uses default i.s.o. user defined layout.

PietKnot, 2022-05-23:
on Android using a smaal screen smartphone, the user may define it's selection widgets and positions in the panels.
[image]
The actual selection made in the editor:
[image]
This is what's presented (nothing)
[image]
Activating also the default 
[image]
Result is the default layout is presented i.s.o. the user defined small layout  . . . . nasty.
[image]
Seems to be a small bug.

wellenvogel, 2022-05-23:
Not sure if I understand the problem correctly.
If you select "small" in the layout editor you are defining the layout that is used if the "small" condition is met.
The setting for the"small" condition is portrait layout below(px).
So maybe it would be better to rename small to "portrait" -- at least for the display.

PietKnot, 2022-05-23:
I assume a miss-understanding on both sides. First of all, in the layout-editor, the current condition is small without any user change and i have only one setting defined at the level of starting the Layout-Editor for user "Osprey". 

Than in portrait mode i did make a configuration and tested oke.
[image]

switched to landscape mode by changing the position of the phone entered the layout editor and defined the "landscape"-configuration the same as de "portrait"-configuration. This one is not displayed but the "default" configuration (see start-of-thread)

I dont understand your remark: "the setting for the"small" condition is portrait layout below(px).
So maybe it would be better to rename small to "portrait" -- at least for the display." What is "(px)" and where / what to rename to portrait or landscape?? I'm starting the layout-editor always with the same filename for user-field "Osprey" and assume that AvNav maintains internally a portrait AND a landscape configuration for the two positions within the file for user "Osprey"

Probably my assumption is wrong, and i have to make two users: Osprey-portrait and Osprey-landscape correct? Will try these options.

wellenvogel, 2022-05-23:
Ok, so I wait for your further tries...
I understand that it somehow could be confusing.
Basically each layout can have different widgets for displays with a width >= portrait width and for displays < portrait width.
Which of the 2 sets you are editing currently depends on the "small" flag.
This is set initially when you start editing by the current situation.
So you can edit for portrait mode even if you rotate later to landscape - as long as "small" is ticked.
You can easily check the 2 parts of the layout by ticking/unticking small without rotating the device.
And to make confusion perfekt:
If your device width is bigger then portrait pixel even in portrait mode - you will never get what you edited for "small".

PietKnot, 2022-05-23:
Looks like i've got it: confusing naming and not reading the manual for checking the intentions.
1. small versus narrow: miss-understood the wording "small"; suggestion to replace to "portrait". At the same time default rename it to "landscape" Also a change in the msg "starting the layout editor" could help the AvNav newbees  by the extended msg "starting the layout editor in Portrait / Landscape" mode. 
2. my SMALL screen resolution is 1080 x 540 while the "portrait layout below (px) " settings was default 480 and quite low,; thus stayed in one mode: "AvNav-landscape"  mode while i thought it was in "portrait" mode.  Now settings changed to 600 for further testing. Possible increase the default to more  modern screen-resolutions.

Checked the operation and all fine now, thus up-to-designer to take above into account to reduce possible confusion when not reading / checking the extensive manual for details.

wellenvogel, 2022-05-23:
Anyway thanks for reporting it.
I'm finally always somehow locked to my thoughts - so it is helpful to see where problems arise.
Will try to improve the doc. Keeping it open to be reminded...

closed 2025-07-23 by wellenvogel