wellenvogel/avnav #228: Userapp does not open with the right url

Fietsenwiel, 2022-05-24:
All userapp opens as intended but one doesn't.
Somehow the url is changed while opening, see screenshots.
[image]

All userapp work accept the last one. When I open this one the url $HOST:9091/demo is changed in $HOST:8080/demo   :

[image]

The url $HOST:9091/demo is a valid url:
[image]

I don't see the difference with the other userapps except I open the url in a seperate window. But that is not the problem, I have checked

wellenvogel, 2022-05-24:
Could you please attach your current avnav_server.xml?

wellenvogel, 2022-05-24:
Ok, there are some smaller bugs in the addon handling. Basically it forgets new window and it reverts $HOME to the current host when editing.
But this does not explain the issue.
Does this also occur if you do not use "new Window"?
And: Which browser do you use?

Fietsenwiel, 2022-05-25:
You say: "Basically it forgets new window and it reverts $HOME to the current host when editing.".

But for the other 4  userapps it works perfedtly well! Only the last userapp (elog) it does not.
It doen not make any difference if I don't use "New window".
And when I change $HOST to 10.8.0.3 it also goes wrong with this userapp. The port is changed to 8080 in stead of 9091

wellenvogel, 2022-05-25:
Very strange. I exactly used your config and it directly tries to open 9091.
So we have to test further.
(1) could you disable the app that runs on 9091 and try to open the page again from AvNav - we should now see an error of course
(2) could you open the development tools on your browser (network tab) shortly before you open the page with the user apps in AvNav. Then open the page that shows all the user apps. In the list there should be a request:
```
avnav_navi.php?request=list&type=addon
```
Could you just attach the response for this request. This will help us to find out if the WebApp gets the correct info for the user apps.

Fietsenwiel, 2022-05-27:
I think it has somthing to do with the application I try to open.

When I disable the app that runs 9091 I get:
[image]

So Avnav does that well.

And I now found out that when I then start the app that runs 9091 and I click on "opnieuw laden" then the page opens with port 8080, very strange.

So Its not a problem in AVNav I guess.

wellenvogel, 2022-05-27:
Good to know...
Anyway it was a nice trigger to find the 2 small bugs...
Will close this once my bug fixes are part of a release.

Fietsenwiel, 2022-07-07:
Closed, problem was not in Avnav. The url needed an extra "/" behind it.

closed 2022-07-07 by Fietsenwiel