wellenvogel/avnav #229: Improve backup / restore a bit

PietKnot, 2022-05-28:
when designing/testing/using AvNAv on different devices (here Windows / Linux / Android-Nexus / Andoid-Phone in use) it's some work to synchronize the status for a user.

Today, the settings.json & layout.json & user.js & user.css & ..  have to be individually copied from differents tabs / restored to different tabs . Would be handy to Export-all / Import-all  (in one zip) somehow to have backup's etc.

Additional settings to be stored for Android:
- working directory (make external also default)
- show demo charts
- hide system bars

Above may help when updates are done on the daily demo-builds (only experience from official to daily yet). Please reject/close if it's more than minor work, as users can do this also with current means.

wellenvogel, 2022-05-28:
For sure a good idea.
Basically it somehow calls for a backup/restore feature.
Packing a couple of files together is not that much of a problem. But you need to carefully select which ones. Most probably you also need pictures, routes, tracks, charts?...
And the import side is also at least not trivial - most probably you need to give the user a chance to decide about overwriting files and potentially selecting the files you would like to import.
Anyway the issue is a good marker.

PietKnot, 2022-05-28:
indeed charts, / routes / tracks / images / keys etc. could be a candidate also, but i would vote -for simplicitely- just those files that can be considered as : "configuration files for AvNav". All others are outside the scope i was thinking of:  e.g. charts, can de GBytes (in my case), and are not part of the AvNav configuration as such; saving them by a simple directory copy is fine enough.  However,  i consider an external Chart directory definition as a configuration item. 

Up to you to decide . . . . i'm happy as is and just thinking in the direction of a small improvement..

The simple widgets / dashboard for my personal use are now ready and synced over the devices and ready for practical experience on usability.

antonical, 2022-09-30:
Hello,

I think a backup and restore feature would be a great and I believe required addition to the functionality of the server. If you look at the approach of something like openwrt then it's really the config of the server we are looking to backup. In our case that probably needs to include routes and tracks and any customisation that may have occurred. But I think chart files are out of scope due to size/time etc.

The objective is to restore the server to its former configuration in the event of failure of the primary disk of the server the hardware etc. Also useful for testing different configs.

Currently, we are imaging the entire server to our onboard NAS. We are also looking at mapping the local storage to network folders for all system files/charts etc.

So the AVNAV server is really exactly that. 

Cheers
Tony