wellenvogel/avnav #230: VMG calculation seems questionable

PietKnot, 2022-06-05:
i noted that the VNG calculation result in a far too low value. When the COG and BRG to the next waypoint (even to the end-of-route) are almost identical the COS-value for the diffenerence should be around "1". 
[image]
[image]

Doing a calculation i assume that the COS-function is implemented on degrees i.s.o. radian
[image]

Implementing the VMG-widget, i used the storevalue:
var vmgrgWidget = {
	name: "userVMGgauge",
	storeKeys: {value: 'nav.wp.vmg'}, //direct in props.value
	caption: "VMG", 
	unit: "Knts",
	type: "radialGauge",

Possibly the widget is wrong or the VMG-storevalue has to be used differently. . . . . .

PietKnot, 2022-06-08:
Guess that the internal presentation is m/sec thus have multiply the storekey value with about 2 to get to knots.

closed 2022-06-08 by PietKnot

wellenvogel, 2022-06-08:
Correct. Internal values are always m/s. You could use formatSpeed...