wellenvogel/avnav #234: Add a digital logbook on avnav

Surfer2010, 2022-06-08:
Hello,
at the Forum we discuss if there is a need or big interest about adding the functionality of a digital logbook to avnav since most of the data is allready present. The additional data like anchor up/down, engine on/off ... could be added by button presses and some plain text addition.

Here the [Link](https://www.segeln-forum.de/thread/83917-t%C3%B6rn-vorbereitung-dokumentation-und-nachbereitung/)  to the discussion on the german forum

Please vote

PietKnot, 2022-06-10:
don't know how to vote, but no interest from me in such a function in AvNav  . . .

Fietsenwiel, 2022-06-13:
You could elog as an alternative.
It has an api so you can log anything you want from scripts...

antonical, 2022-09-30:
I think the idea of a digital logbook is a valid one and would love to see this functionality in AVNAV. Today we use OpenCPN plugins to do this.

Having it natively in the AVNAV server would be excellent. 

Cheers
Tony

antonical, 2023-07-06:
I would also love to see the addition of a digital logbook recording data at periodic intervals.  It would also be great to have a check box or button to signal engine on/off which would then be recorded in the log along with Time/Position/Active Route/COG/SOG/HDG/WIND etc.

gpenot, 2025-06-21:
Important for me, I love AVNAV but I can't drop définitively Openplotter because there is no loogbook.