wellenvogel/avnav #240: Enhancement Layout Editor Dashboard

PietKnot, 2022-07-05:
With current screen formats of small and large smartphones which have often an aspect ratio height/wide of about 18:9, the design of the dashboard becomes difficult especially using Gauges; to be readable, only one gauge on the top is possible and values below create a lot of white space. This request is to enhance the layout editor by either:

a. add a top-panel(configurable height) to the existing Left/ M 1,2,3 /Right panels 

b. allow 2 or 3 widgets ("combine") in one panel-colomn horizontal side-by-side (thus when adding not only above/under, but also left/right).

My preference would be option a. as it is inline with the map-layout and possibly easier to implement. Option b. would give more -maybe not needed- flexibility, and also usable for more standard aspect ratios like used in tablets, etc.

Further, allow easy configuration of the header-line of the widget-boxes size/color (e.g. dark-blue background with white font); possibly just a documentation issue as with try-and-error not found with designer tools.

An example of above white space "problem":

[image]

where using the same type op widgets on a Map, this white space problem does not occur:
[image]

wellenvogel, 2024-09-22:
With the daily from [20240914](https://www.wellenvogel.net/software/avnav/downloads/daily/20240914/) you can now use a "CombinedWidget" to add multiple widgets to one "slot" in a column on the dashboard (and on the nav pages). You can arrange them in a row or in a column.
You could even nest multiple CombinedWidgets - although editing could get a bit hard.
You can use the normal editing- or now also freely move the widgets around with drag and drop.
Would be interesting to know if this addresses your issue.

PietKnot, 2024-09-25:
According to the description, this solves my "issue". In the mean time, i got used to the limitation in porttrait view, and will test it in landscape. Thus issue closed with thnx!!

closed 2024-09-25 by PietKnot