wellenvogel/avnav #246: First page with chart selection is very confusing for a user

mgrouch, 2022-09-13:
First page with chart selection is very confusing for a user

There is lots of abbreviations
mp-base
mp-noaa
mp-bsh
mp-wsv
noaa-enc-online

First noaa, bsh, wsv cover different regions. So why can't they all be 'on' together

'mp-noaa' does it exist? it's not an on-line service anymore. There might be offline way to download raster mbtiles for NOAA but
mp I guess assumes they are on-line

Thanks

wellenvogel, 2022-09-13:
Would be really helpfull to describe the problem in more detail.
This includes  versions of used software, configs etc.
For the "confusing first page":
AvNav currently handles chart sets and will display one of them at a time ( until you define overlays).
So no easy way to merge all the mapproxy sets without hiding the source to the user.
mp-noaa is still there for users having caches of it.
mp-noaaenc is the new online service.
noaa-enc-online is not there at all.
For SK charts:
Please provide an example.
I do not have time to install AvNav on any possible system - so to address problems I need the details.

mgrouch, 2022-09-13:
Easy to reproduce by installing latest BBN Marine OS image: https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html#download

Thanks

wellenvogel, 2025-07-23:
closing this due to no further response

closed 2025-07-23 by wellenvogel