wellenvogel/avnav #248: log file filling all the memory

romton843, 2022-09-30:
Hi, 
The log file filled up 8GB of memory in 6 hours making up to the memory was full.
It writes an error up to 6 times per milliseconds, here it is:

2022-09-29 22:47:54,534-724-1307-Plugin: builtin-Sail_Instrument-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2022-09-29 22:47:54,535-724-1307-Plugin: builtin-Sail_Instrument-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2022-09-29 22:47:54,535-724-1307-Plugin: builtin-Sail_Instrument-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2022-09-29 22:47:54,535-724-1307-Plugin: builtin-Sail_Instrument-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2022-09-29 22:47:54,535-724-1307-Plugin: builtin-Sail_Instrument-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside

Do you know if there a way to limit the log file size or to fix this error or to write it maybe once per minutes or seconds or just once?
Best regards.
Romain

free-x, 2022-09-30:
- Open issue at https://github.com/kdschmidt1/Sail_Instrument/issues
- You can uninstall plugin
- You can disable plugin
Greetings
free-x

romton843, 2022-09-30:
Thank you. I'll try that.

kdschmidt1, 2022-09-30:
Hello,
A first investigation is indicating: (Please skip if you are not interested in details)
The Error is emitted by Avnav, not Sailinstrument Plugin, caused by a call to AVNAV-API-function getDataByPrefix() from Sailinstrument.
This API function Error is indicating corrupted data in the store. To solve this, it would be mandatory to investigate the root cause for data corruption together with andreas.

It would be helpful, to see your server data. Can you please try the follwing URL in your browser and share the result?
->[YOUR AVNAV IP]/viewer/avnav_navi.php?request=gps
Anyway i will try a workaround for Sailinstrument this weekend by replacing the  getDataByPrefix() call with getSingleValue() calls, hoping this helps. I will come back with updates.
regards klaus

wellenvogel, 2022-09-30:
Thanks for reporting.
Basically AvNav should limit it's log. So I will cross check again.
Btw: Which version of AvNav are you using?

kdschmidt1, 2022-10-02:
A new release of [SailInstrument](https://github.com/kdschmidt1/Sail_Instrument) with the proposed workaround for this issue is now available. Would be nice if you can check if this solves the problem.
regards klaus

free-x, 2022-10-02:
FYI sailinstrument 20221002 is now available in "debpreview" channel/repository

Greetings
free-x

romton843, 2022-10-02:
I'm using AvNav 20220819 with avnav-sailinstrument-plugin 20220615.
I'll wait for the plugin update to be available on AvNav.
Here is what returns /viewer/avnav_navi.php?request=gps
The wind instrument is not connected, could it be what's causing the issue?

updatealarm | 1664730489.716585
updateleg | 1664730490.716667
updateroute | 1664730489.7167
updateconfig | 1664730489.716731
version | "20220819"
signalk |  
environment |  
outside |  
temperature | 299.09
pressure | 101402
inside |  
humidity | 60.876
gas | 542847.2581757255
airquality | 246
rpi |  
gpu |  
temperature | 327.15
cpu |  
temperature | 327.68
utilisation | 0.28
core |  
1 |  
utilisation | 0.05
2 |  
utilisation | 0.43
3 |  
utilisation | 0.6
4 |  
utilisation | 0.04
memory |  
utilisation | 0.24
sd |  
utilisation | 0.85
observations |  
stationName | "Charleston, Charleston Air Force Base"
stationId | "KCHS"
elevation | 15
temperature | 22.8
dewpoint | 13.3
windDirection | 350
windSpeed | 16.56
barometricPressure | 101250
seaLevelPressure | 101250
visibility | 16090
relativeHumidity | 55.037015331205
heatIndex | 22.57263317809278
moon |  
fraction | "0.47"
phase | "0.24"
phaseName | "Waxing Crescent"
angle | "-1.61"
times |  
rise | "2022-10-02T18:21:31.335Z"
alwaysUp | "false"
alwaysDown | "false"
sun | "day"
mode | "day"
sunlight |  
times |  
sunrise | "2022-10-02T11:15:45.132Z"
sunriseEnd | "2022-10-02T11:18:17.696Z"
goldenHourEnd | "2022-10-02T11:48:27.571Z"
solarNoon | "2022-10-02T17:10:23.786Z"
goldenHour | "2022-10-02T22:32:20.000Z"
sunsetStart | "2022-10-02T23:02:29.875Z"
sunset | "2022-10-02T23:05:02.439Z"
dusk | "2022-10-02T23:29:39.135Z"
nauticalDusk | "2022-10-02T23:58:13.431Z"
night | "2022-10-03T00:26:55.003Z"
nadir | "2022-10-02T05:10:23.786Z"
nightEnd | "2022-10-02T09:53:52.568Z"
nauticalDawn | "2022-10-02T10:22:34.140Z"
dawn | "2022-10-02T10:51:08.436Z"
name | "Romanopi"
navigation |  
courseOverGroundTrue | 4.536110726968991
courseOverGroundMagnetic | 4.67783146223093
magneticVariation | -0.14172073526193954
position |  
longitude | -79.9589
latitude | 32.81255
speedOverGround | 0.04630001172933631
magneticVariationAgeOfService | 1664739250
datetime | "2022-10-02T19:34:10.000Z"
headingTrue | 5.593780253919052
track | 259.90000005934286
lon | -79.9589
lat | 32.81255
speed | 0.04630001172933631
time | "2022-10-02T19:34:10.000Z"
MagVar | -8.122542518972875
headingTrue | 320.50000007317965

Tanks everybody.
Regards.
Romain

Fietsenwiel, 2022-10-07:
I have the same problem.
How can I update this plugin. 
There is nothing in /home/pi/avnav/data/plugings. But I see the plugin in "AvNav update" and refreshing the packageinfo does not present a new candidate for this plugin.....

free-x, 2022-10-07:
you have two options
1. wget https://github.com/kdschmidt1/Sail_Instrument/releases/download/20221002/avnav-sailinstrument-plugin_20221002_all.deb ; sudo dpkg -i avnav-sailinstrument-plugin_20221002_all.deb
2. activate daily repository in your sources.list
deb [signed-by=/usr/share/keyrings/oss.boating.gpg] https://www.free-x.de/debpreview bullseye main contrib non-free

wellenvogel, 2023-04-29:
On AvNav side the error messages have been corrected.
closed with 20230426

closed 2023-04-29 by wellenvogel