wellenvogel/avnav #253: navigation signalk to nmea2000

mat214, 2022-11-02:
Hello,

After a season of using avnav, I find the client/server principle at the top.

I am modernizing the electronics of my small sailboat with an n2k network. Avnav communicates with an I70s screen.

Avnav and signalK communicate together with AVNSignalKHandle.

It works fine except that I can't send navigation information to the I70s screen. the signalk-to-n2k plugin uses the patch: navigation.courseRhumbline.bearingToDestinationTrue

While avnav writes the patch: navigation.courseGreatCircle.nextPoint.bearingTrue

As a result, the navigation information is not sent to the n2k network...

Is it possible to replace in the naming of the patches "GreatCircle" by "Rhumbline"

Thank you for your help and especially for the work around avnav!

wellenvogel, 2022-11-03:
Just switch AvNav routing mode to rhumbLine
https://wellenvogel.de/software/avnav/docs/quickstart.html?lang=en#h2:Routes

mat214, 2022-11-03:
But that doesn't completely solve my problem.

Via signalk to communicate avnav. Signalk does not return n2K info...

When I enable writing in nmea 0183 between avnav and signalK, then signalk writes the info in n2k. the DTW is showing on the screen but I have 000 for the BTW.

strange because in both cases the signalk patch exists but in one of the configurations the log of the signalk-to-n2K plugin does not return anything from avnav ...

wellenvogel, 2022-11-03:
Very strange.
But for me it's hard to tell what is going on there as I have no real idea what the plugin is doing.
Maybe I will have a look - but will take some time.

mat214, 2022-11-04:
Yes it is strange. I feel like the n2K plugin only returns the nmea's and not the signalk data...

I installed openplotter. The installation configures for avnav a fallback to signalk in nmea for APB and RMB.

Doing this returns it works. To get the info BTW on my i70s screen I also had to activate computeAPB in AVNrouter.

The XTE is missing. I have the impression that AVNAV does not review the XTE sentence in nmea.

Thanks again for the software!

wellenvogel, 2022-11-10:
AvNav currently does not generate XTE (it's included in RMB...).

wellenvogel, 2024-05-29:
Could you maybe re-test with the latest version?
I included a workaround at least for some SignalK bugs to write some information into multiple pathes.
See https://github.com/wellenvogel/avnav/blob/6242aebf90aaf6f8d55a3f5628680328f2e07824/server/handler/signalkhandler.py#L1428

mat214, 2024-06-02:
I will be sure to report back with the latest update. Unfortunately it won't be long before I return to my boat in over a month.