wellenvogel/avnav #254: Copyright Violation

femski, 2022-11-15:
Hello, I am the creator of AvNav - a flight navigation App for pilots widely used since 2014. AvNav is our registered trademark (we also own the domain name). [Please see here for details](https://tmsearch.uspto.gov/bin/showfield?f=doc&state=4801:u7qgn7.2.1)

I see you are using the same name for your project/product - this steals traffic away for our product is causes confusion and harm. Kindly consider another name for your project. I would like to avoid filing formal complaint and hassles it may cause you and us. thank you,

wellenvogel, 2022-11-15:
I will have a look.
My software and app is not used commercially at all - and I have to clearify the impact.

wellenvogel, 2022-11-16:
I had some deeper look into the issue.
My findings:
(1) I don't think that my usage of the term "AvNav" does violate your rights at all as it is completely non-commercial.
(2) AvNav (on my side) was out there long before you registered your mark
(3) Your registration is US only - and is restricted to the following areas:
`Computer software for flight planning and navigation for pilots; Mobile application for flight planning and navigation for pilots`
`Software as a service (SAAS) services, namely, hosting software for use by others for use in flight planning and navigation for pilots`
My AvNav does not touch any of those areas - it's completely focused on navigation for boaters.

On the other side I also do not like the idea of users being confused. I already had complains of users that have been landing on your app while searching for mine in the store.

So I'm thinking about some compromise.
I could imagine to make some change in the name for my Software (trying to get closer to the main focus).
But I would limit this to the title in the Play Store (and in the app) and to the wording in the documentation.
Everything else (like package names for linux, repositories and similar) would remain as they are.

If you are interested we could additionally go for some cross-linking to guide the users where they really would like to go (i.e. link to your company on my landing page/store entry and vice versa).

femski, 2022-11-17:
Thank you - if you change the title of the software - in Play Store, product documentation and any promotional website - then that's seems like an acceptable resolution to me. You can keep package name, name of Github repository etc. I understand its pain to change those involving much work and they are of interest to software developers only - who know better.

Top links with AvNav in title show as:
[https://open-boat-projects.org/en/avnav/](https://open-boat-projects.org/en/avnav/)

[https://www.wellenvogel.net/software/avnav/docs/beschreibung.html?lang=en](https://www.wellenvogel.net/software/avnav/docs/beschreibung.html?lang=en)

[https://play.google.com/store/apps/details?id=de.wellenvogel.avnav.main&hl=en_US&gl=US](https://play.google.com/store/apps/details?id=de.wellenvogel.avnav.main&hl=en_US&gl=US)

If you change the title of your App - I presume these will change to reflect the new title.

I reached out at this stage as we are soon launching a revamp of our product and the website and I wanted to see that confusion was eliminated Thank you for your cooperation.

wellenvogel, 2022-11-17:
Ok, thanks for the reply. I guess we are making progress.
But before starting any changes I would like to clearify a couple of things:
(1) As stated in my previous comment I do not see any legal obligations that will REQUIRE me to make any changes. As you did not comment on this I assume that you agree.
(2)It is not my intention at all to create some negative impact on your business. I understand that you need to get money out of it. And I also do not want to confuse users/customers.
(3) Therefore my proposal ( that creates reasonable effort on our side).
Anyway I don't think that my name changes will solve your issue of google ranking. First of all the new name will have to stay rather close to the existing one (maybe something like AvBoatNav) - users need to have a chance to get the relation.
We are just discussing about a name in the community.
On the other hand it's a community project - so people are discussing about it, asking questions, making videos. I have no influence on the naming that they will use (and for sure they will still use "AvNav" for some time). Not sure how google will rank this.
As I wrote - my idea would be cross linking. But it's up to you to decide.

femski, 2022-11-17:
I can not say if you are obligated or not - I am not a legal expert. I think you should. I think its moot point debating it since I am not involving lawyers and not seeking any damages or further action - if we can amicably resolve this. We seem to have at least an outline for what that might be.

But AvBoatNav - to me does not set it apart enough - its still mostly AvNav - besides why would you want Av anything (usually meant for Aviation businesses) in a boating application name? we get asked about writing a boating App it and I can tell you if we ever did it - we wont call it AvNav or Av anything.

I would encourage community to think of a more appealing and unique name. Something like Wellengovel BoatNav would be distinct and clear retaining major part of its old name. But its up to you.

if you change the name then other sites will follow - and we wont need cross linking either. so i don't see any need for changing anything besides name of your App at this point. Thank you

wellenvogel, 2022-12-02:
Basically the whole process of changing the name is not that easy.
As we have other stuff that belongs to AvNav (commands, debian packages, plugins, videos on YT, external write ups,...) there is clearly a must to remain relatively close to the well known name.
Currently I triggered some voting in the community for name proposals.
On the other hand if it will not really help you if we make some changes like the one I proposed I do not really see a way forward.
For sure AvNav on our side was out long before you registered your name. And due to the community nature the info is widely spread.
As I wrote - it's not my intention to interfere with your business at all. On the other hand on our site people are just doing everything in their spare time - so resources are really limited. Therefore I was looking for a compromise that is acceptable for  both sides. 
But I'm not really sure on how much you agree on the compromise idea (i.e. what is the compromise part on your side?).

FredericGuilbault, 2023-01-25:
I think the Canadian coast guard had it first, It's the name of the service that reporting missing buoys and publish charts corrections.  ....Since 1867 they say on their website. :)

https://www.ccg-gcc.gc.ca/mcts-sctm/navwarn-avnav-ca-fra.html

wellenvogel, 2023-01-25:
Interesting point, thanks for bringing this up.

wellenvogel, 2024-02-19:
No further activity - so closing this.