wellenvogel/avnav #255: Start in split screen

captainfast-git, 2022-11-20:
Hallo Andreas,
the split screen function is so great! It would be nice, to allow the app to start in split screen with the last setup.
Maybe to have the possibility to enable the "start in splitscreen" function as an option in settings.

Thanks

wellenvogel, 2022-11-24:
Ok, not it the settings right now...
Not sure which App you are addressing - if you just use the browser-access you can bookmark the split screen url and use this one.
http://somehost:nnn/viewer/viewer_split.html.
Having this in the settings is not that easy as each part of the split screen view would read the settings. But I will think about it...

wellenvogel, 2024-06-19:
20240616

closed 2024-06-19 by wellenvogel