wellenvogel/avnav #256: Chart groups and scale

thefrenchsailor, 2022-12-03:
I tried importing BSB chart folders containing hundreds of charts each but it seems to only work with individual charts files. Note that the chart folders may have sub-folders depending on the source. Maybe it could be done by importing a compressed .zip file directly and letting AvNav handle the conversions.

Besides importing, I also can't group them together to show at the same time. This could be done by using the folder structure of the chart directory to determine groups. For instance, if there's a folder in the directory, it is automatically a group and all charts inside are shown together.

It would also be good to detect or specify the chart scale from the individual charts to show/hide them dynamically when navigating as some charts may be more or less detailed for certain zoom levels. For example, when I zoom in, it would hide the larger charts and instead show the more detailed ones. Vice versa.

Therefore, I ask;
- importing .zip containing multiple charts
- generating chart groups based on their directory
- showing/hiding individual charts based on their scale

I hope this I am explaining myself adequately, thank you.

thefrenchsailor, 2022-12-03:
After further research, I have partially answered my questions.

It seems the windows converter can be used with chart dataset folders. 

Yet;
- can the windows converter handle zoom layers?
- is there a converter on the OpenPlotter version of AvNav?

Thank you.

free-x, 2022-12-03:
> can the windows converter handle zoom layers?
yes

> is there a converter on the OpenPlotter version of AvNav?
Yes, is the same "engine" as under windows, only without UI. On command line
`/usr/lib/avnav/chartconvert/read_charts.py -g -m all -o <name_of_outputchart>  /path/to/bsb/`

thefrenchsailor, 2022-12-03:
Very well then!

I will give it a try and report back if issues are encountered.

Thank you free-x.

closed 2022-12-03 by thefrenchsailor