wellenvogel/avnav #257: Sub directories in charts

dakshsingh, 2022-12-23:
I just installed avnav and am trying to get it to access my collection of mbtiles files. They are arranged in country-wise directories. I copied the entire collection in their folders and placed it in /home/pi/avnav/data/charts

I discovered that avnav can only see the mbtiles files in the charts directory itself. It doesn't read any mbtiles files in the subdirectories. Ideally in the home page I'd like to see the entire directory structure to navigate through and then select the mbtile file to be used. 

On the chart list page I also see a number of charts named mp-base, mp-bsh and so on. None of these are relevant to me. Is there any way to remove these?

I have the latest version installed on openplotter 3

wellenvogel, 2022-12-23:
Unfortunately subdirs are not handled right now.
Maybe you could think about merging some mbtiles into larger ones. Mbtiles is a rather efficient format - so bigger files are no problem.
The mp-xx charts are from the mapproxy plugin.
Just disable the plugin on the server/status page at the plugin manager.

dakshsingh, 2022-12-23:
Thanks for the quick response. 
Unfortunately each of my mbtiles files is already several GB as it they go down to very granular zoom levels.

I think having a way to add directories in the interface for collections of charts and a browser to navigate through the directories will be very useful. One can then share the charts repository between multiple charting software including reusing OpenCPN chart collections that one may have built up. May be an idea for an enhancement.

Thanks for the tip on mapproxy. That worked nicely.

wellenvogel, 2022-12-23:
Enhancement ideas are always welcome ;-)