wellenvogel/avnav #261: Log file filling all the memory again

romton843, 2023-02-09:
Hi,
Could you please do something to prevent Avnav to fill up 8Gb of memory in few hours with log files?
Like an option in the settings to prevent the log to become too big or to be able to disable the log file.
Please do something.
Here a split second of my log.
Best regards.
Romain

2023-02-09 17:30:34,440-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,441-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,441-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,441-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,441-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,442-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,442-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,442-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,442-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,442-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,443-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,443-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,443-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,443-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,443-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,444-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,444-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,444-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,445-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,445-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside
2023-02-09 17:30:34,445-718-1062-TrackWriter-ERROR-inconsistent data , found normal value and dict with key signalk.environment.outside

wellenvogel, 2023-02-10:
Could you please provide some information about the versions you are using?

wellenvogel, 2023-02-10:
could you please provide the output of:
`curl 'http://nnnn:3000/signalk/v1/api/vessels/self'`
With nnnn being the ip address of your system (localhost if you are on the system itself).

romton843, 2023-02-10:
20220819
Everything is up to date.

romton843, 2023-02-10:
{"uuid":"urn:mrn:signalk:uuid:1c1e42a2-77b4-4c12-a409-130eacaa7b3f","name":"Romanopi","environment":{"forecast":{"description":{"meta":{"timeout":3600},"value":"connecting to openweathermap...","$source":"openweather-signalk","timestamp":"2023-02-09T23:08:13.402Z"}},"rpi":{"gpu":{"temperature":{"meta":{"units":"K"},"value":324.25,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:30.881Z"}},"cpu":{"temperature":{"meta":{"units":"K"},"value":324.76,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:30.884Z"},"utilisation":{"meta":{"units":"ratio"},"value":0.32,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:05.836Z"},"core":{"1":{"utilisation":{"meta":{},"value":0.06,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:05.837Z"}},"2":{"utilisation":{"meta":{},"value":0.05,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:05.838Z"}},"3":{"utilisation":{"meta":{},"value":0.22,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:05.839Z"}},"4":{"utilisation":{"meta":{},"value":0.96,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:05.839Z"}}}},"memory":{"utilisation":{"meta":{"units":"ratio"},"value":0.28,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:30.885Z"}},"sd":{"utilisation":{"meta":{"units":"ratio"},"value":0.69,"$source":"signalk-rpi-monitor","timestamp":"2023-02-10T13:16:30.893Z"}}},"outside":{"temperature":{"meta":{"units":"K","description":"Current outside air temperature"},"value":296.59,"$source":"signalk-raspberry-pi-bme680.XX","timestamp":"2023-02-10T13:16:31.135Z"},"pressure":{"meta":{"units":"Pa","description":"Current outside air ambient pressure"},"value":102168,"$source":"signalk-raspberry-pi-bme680.XX","timestamp":"2023-02-10T13:16:31.135Z","trend":{"tendency":{"meta":{},"value":"RISING","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"trend":{"meta":{},"value":"CHANGING","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"severity":{"meta":{},"value":2,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"period":{"meta":{"units":"s","type":"number","description":"Trend period, 60 or 180 minutes ","displayName":"Trend period","shortName":"Trend period","example":3600},"value":10800,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z","from":{"meta":{"units":"Pa","type":"number","description":"Trend period start pressure","displayName":"Period from pressure","shortName":"Trend from","example":101325},"value":101976,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"to":{"meta":{"units":"Pa","type":"number","description":"Trend period end pressure","displayName":"Period to pressure","shortName":"Trend to","example":101325},"value":102168,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"}}},"prediction":{"pressureOnly":{"meta":{},"value":"Expect fair, dry, cool weather and a strong breeze","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"quadrant":{"meta":{},"value":"N/A - No true wind data","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"season":{"meta":{},"value":"Fair, windy, much cooler","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"beaufort":{"meta":{},"value":"Less than F6","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z","description":{"meta":{},"value":"Less than a strong breeze","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"}},"front":{"tendency":{"meta":{},"value":"Continously rising","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"prognose":{"meta":{},"value":"Post cold front","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"wind":{"meta":{},"value":"Gusty and easing","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"}}},"system":{"meta":{},"value":"Between Low and High","$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"1hr":{"meta":{"units":"Pa","type":"number","description":"Pressure 1 hour ago","displayName":"Pressure 1 hour ago","shortName":"Pressure 1 hour ago","example":101325},"value":102116,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"3hr":{"meta":{"units":"Pa","type":"number","description":"Pressure 3 hours ago","displayName":"Pressure 3 hours ago","shortName":"Pressure 3 hours ago","example":101325},"value":101974,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"6hr":{"meta":{"units":"Pa","type":"number","description":"Pressure 6 hours ago","displayName":"Pressure 6 hours ago","shortName":"Pressure 6 hours ago","example":101325},"value":102066,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"12hr":{"meta":{"units":"Pa","type":"number","description":"Pressure 12 hours ago","displayName":"Pressure 12 hours ago","shortName":"Pressure 12 hours ago","example":101325},"value":102198,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"24hr":{"meta":{"units":"Pa","type":"number","description":"Pressure 24 hours ago","displayName":"Pressure 24 hours ago","shortName":"Pressure 24 hours ago","example":101325},"value":102588.00000000001,"$source":"signalk-barometer-trend","timestamp":"2023-02-10T13:16:00.753Z"},"48hr":{"meta":{"units":"Pa","type":"number","description":"Pressure  [... cut ...]  the identifier for the environmental zone, eg. engineRoom, mainCabin, refrigerator"},"value":271,"$source":"signalk-raspberry-pi-bme680.XX","timestamp":"2023-02-10T13:16:31.135Z"}}},"validation":{"I2C":{"meta":{"description":"AC equipment common qualities"},"value":"7b4fc55a-71d9-4a98-a957-b46acc036a57","$source":"OpenPlotter","timestamp":"2023-02-09T23:10:17.231Z"},"GPIO":{"meta":{},"value":"c92a4d9b-0d37-4a61-85a1-ac7b1f8fe140","$source":"OpenPlotter","timestamp":"2023-02-09T23:12:10.497Z"}}}

wellenvogel, 2023-02-10:
Ok, I see.
Basically I agree that the log should not be flooded - I will try to implement a handling for such situations.
On the other hand there is a value in SignalK (vessels/self/environment/outside/pressure/prediction) that does not conform to the schema. Basically there is a value at vessels/self/environment/outside/pressure and there are values at other elements below.
Refer to https://signalk.org/specification/1.7.0/doc/vesselsBranch.html.
Most probably caused by https://github.com/oyve/signalk-barometer-trend.

wellenvogel, 2023-02-10:
OK, triggered a daily version with a correction.
https://www.wellenvogel.net/software/avnav/downloads/daily/20230210/
Maybe you would like to try it out...

wellenvogel, 2023-02-14:
Have you been able to try the daily? Would be interesting to know if it solves the problem - and if creates other issues.

wellenvogel, 2023-04-29:
closed with 20230426

closed 2023-04-29 by wellenvogel