wellenvogel/avnav #264: Formatting valuebox  user Widget gauge

PietKnot, 2023-03-13:
in a user defined widget with "radialGauge" with option "valueBox" false", whis is ignored and in de settingbox of the widget, the option "drawValue" is by default activated and a valye is displayed on the screen. This although no valyeBox was defined.

Personally no problem to deactivate it again, but for new users of this widget (WPD of last year), i get complaints on the value displayed (even with 15 digits). 

How to prevent the formatter to enable the valueBox again, as even the code:

" var wpdrgParameter={
	value: false,
	formatter: false,
	formatterParameters: false"
does not help.

PietKnot, 2023-03-13:
see attached user.js (maybe you remember me from last-year discussing this widget); it's the first user-widget in the attachment. I did not add it, as assumed it was obvious what i like address:

[user.js.txt](https://github.com/wellenvogel/avnav/files/10958911/user.js.txt)

The notion "drawValue" which is being discussed is displayed in the screenshot:

[image]

Hope it is clear now.

PietKnot, 2023-03-13:
sorry, second widget in the attachment.

wellenvogel, 2023-03-13:
Ok, I see.
Unfortunately my description is not correct/sufficient at this point.
The gauge widgets all have a "drawValue" parameter that create an own value display (not using the original valueBox).
So you have to use this parameter.
To be on the save side do 2 things:
(1) add a drawValue:false to your editable parameters - this will prevent the box in the layout editor (or define an own BOOLEAN there with the default of your choice)
(2) if you set drawValue:false in the editable parameters also set drawValue:false in the widget parameters to finally prevent the vale box to be displayed.
I will keep this issue open to adapt the documentation.

PietKnot, 2023-03-13:
confirmed: solution (1)+(2) is working; i thought that i understood a bit AVnav,  but oke; sorry for the disturbance  but when speading a new widget i do encounter these things.

wellenvogel, 2023-04-29:
corrected in doc

closed 2023-04-29 by wellenvogel