wellenvogel/avnav #269: Running in Docker?

tkurki, 2023-04-17:
I'd like to try Avnav "for real" (with o-charts for Finnish waters) but would like to keep it in Docker, like I have all my other software. Do you think it is feasible?

Docker is mentioned in #179 - did you @emilecantin make any progress with it?

This Dockerfile (based on https://github.com/free-x/avnav-gen-arm64)
```
FROM debian:bullseye

COPY files .

RUN apt-get update
RUN apt-get install -y python3-bluez \
python3-pyudev \
python3-serial \
python3-gdal \
python3-netifaces \
python3-pil \
python3-pigpio \
gnupg

RUN apt-key add - < ./oss.boating.gpg.key && \
echo 'deb https://www.free-x.de/debian bullseye main contrib non-free' >/etc/apt/sources.list.d/avnav.list && \
echo '#deb https://open-mind.space/repo/ bullseye-stable avnav' >>/etc/apt/sources.list.d/avnav.list && \
echo '#deb https://open-mind.space/repo/ bullseye-daily avnav' >>/etc/apt/sources.list.d/avnav.list && \
apt-get update
RUN apt-get install -y avnav
ENTRYPOINT [ "/usr/bin/avnav" ]
```

builds and running it gives me a functioning AvNav, at least the osm charts work. 

I guess fiddling with the startup & configuration will make the errors go away and then I could connect to my Signal K server for data.

Is o-charts something that is feasible with this setup?

Is this something somebody else might be interested in?

free-x, 2023-04-17:
hi @tkurki 
at first you need to adjust options for start of avnav
```$ avnav -h
Illegal option -h
invalid option ?
usage: /usr/bin/avnav [-e] [-d] [-q] [-b basedir] [-c chartdir] [-x cfgfile] [-t template] [-g] [-a softwareBase] [-v viewerDir]
       -b basedir     : data dir for avanv routes,tracks,charts,...
       -c chartdir    : separate dir for charts (instead of basedir/charts)
       -x cfgfile     :  start the server directly with this config (not for gui)
       -t template    : create basedir/avnav_server.xml from this if not found
       -l loglevel    : set the log level (ERROR|INFO|DEBUG)
       -e             : directly fail on any errors in avnav_server.xml instead of trying avnav_server.xml.ok
       -d             : start with loglevel debug
       -g             : start with GUI instead of normal server mode
       -q             : do not log to stderr
       -n envFile     : read environment variables from env file if it exists
       -------------- some development options --------------
       -a softwareBase: find the server software at this directory
       -v viewerDir   : find the viewer software at this directory
```
Volume with "data" should be exported

An example of avnav as client of SK can be found at openplotter
https://raw.githubusercontent.com/e-sailing/openplotter-avnav/master/openplotterAvnav/data/avnav_server.xml

Indeed o-charts in docker would be most exciting part

greetings
free-x

wellenvogel, 2023-04-17:
I think it should be possible...
Basically everything has defaults and should more or less start to work out of the box.
As @free-x already said it would make sense to have the data dir (-b) on a mounted volume - but it depends.
Running o-charts should be possible (doing this while developing) - just install the package avnav-ocharts-plugin (it will also install the avnav-ocharts) and the dependencies.
The interesting part is the licensing stuff. I would be rather sure that you could create an FPR and get some charts assigned from the shop.
Not sure about the dongle. But if having the right privileges for USB devices it could work...
Just ensure to have 8080 (AvNav itself) and 8082 (AvNav-ocharts) as ports being mapped (or you need to use the docker network addresses).
For o-charts there could be some issue with the network addresses:
The plugin checks the client address to be in a local network. So potentially using the docker network finally is the only real chance...

For the SK connectivity:
You can easily configure a remote SK host. If you want to "feed" everything from Signalk it should be sufficient to set the correct SK host on the AVNSignalKHandler (Server/Status page), activate "decodeData", "fetchAis". If you want to send back you can activate "sendData". In this case you need to input username/password.
If you want you still can configure other connections - but that's not really necessary for this case.

If you have put your data dir to a mounted volume things should be persistent.
And if you would like to go for multiple instances - just take your adapted avnav_server.xml, put it into the docker image and use it for the -t parameter - this way starting with it as a template.

tkurki, 2023-04-18:
For chart licensing I was thinking about the dongle as the only viable option, I don’t think anything else makes sense. You should be able to recreate Docker containers whenever.

wellenvogel, 2023-04-18:
Agree..
So we just have to find out what the dongle needs - not sure if --device will do or if we have to go for some --privileged mode for the container.
OK - a quick check shows that oexserverd is accessing /dev/bus/usb/** - so not sure if this will work out of the box.

tkurki, 2023-04-19:
Build fails with [this Dockerfile](https://github.com/tkurki/avnav-docker/blob/4c0b8f1fad2878f730d5fa93f2c57d9bc4bf400f/Dockerfile) on RPi 3 arm64
```
#0 10.71 The following packages have unmet dependencies:
#0 11.28  libgtk-3-0 : Depends: libgtk-3-common (>= 3.24.24-4+rpt2) but it is not going to be installed
#0 11.28               Recommends: libgtk-3-bin but it is not going to be installed
#0 11.28               Recommends: librsvg2-common but it is not going to be installed
```

Ideas?

wellenvogel, 2023-04-19:
Additional hint:
As always - when mapping the data dir from outside - take care of user id / permissions.
Basically avnav is able to run with every user. So you could either add a user that has the same id like your host user - or change the userid for avnav...

tkurki, 2023-04-19:
Thanks, I'll worry about userid mapping etc once I get both avnav and avnav-ocharts-plugin to install & start.

Not sure how much time I'll have for this in the coming weeks, but at least I have a start that even somebody else could continue if I fall by the wayside...

wellenvogel, 2025-08-15:
For AvNav it should work.
But: Why don't you simply use the packages?
The build command is nothing being stable.
At the end you only need the avnav package (and some plugins if you want).
And this will install all depenencies as needed.
For o-charts things are more complex.
In principle it will work but the fingerprint is bound to a system.
Not sure if this will be stable when using docker.

tkurki, 2025-08-15:
My motivation for using Docker in general is the ”containerization”:
- everything needed, with matching dependencies that work together, are in the same package
- ease of updates
- ease of uninstall
- ease of testing different versions and combinations of different services
- one more layer of security

In Docker the hardware key is really the only feasible option.

free-x, 2025-08-15:
I have always created Docker container for AvNav with all possible plugins ( mapproxy, ochartsng).
https://github.com/free-x/avnav-docker

Docker available for dailies ( https://hub.docker.com/r/xfreex/avnav-daily)  , but I'm planning to push to hub also release docker.

Greetings
free-x

stackshadow, 2025-08-15:
Well i coming from nixos, i am trained to use the sources for build-stuff 🐉 😄 ( and even just can create a vm from nix-files, maybe this will help according to the fingerprinting 🤔 )

But yeah, no need to create a second docker build if free-x already has an working state :)
I will grab this dockerfile and play around with the fingerprinting that can be generated from the ui, maybe found something that the fingerprint always the same 😺 

Just some questions according to fingerprinting:
Is it documented how the fingerprinting works ? For example do we know if it uses the mac/cpu-id or something ?
Is the fingerprinting a thing when we download the charts with the online-function of avnav ?

wellenvogel, 2025-08-15:
The fingerprinting is important to bind the charts to the dedicated device.
So for licensing issues the process on how the fingerprint is created is secret.
And I will not support any attempt to analyze/modify this. From my pov o-charts is really making a good job in providing official charts for open source software for a reasonable price - so I would avoid anything to create trouble to them (as they have to work with all the chart issuers and have to prove that they have a working licensing).
I understand the idea of running everything in docker (I also like it...) - but I also suspect the dongle would be the only working solution (and this would still need to have the USB devices mapped to the container).

Btw: I feel the docker build by it's own is rather straight forward - just add the repos, install the packages and create a run command that simply starts AvNav.
Things are getting more complex if you think about the usage:
(1) You need to map a location to store all the data (AvNav's data dir) - this by nature would be outside the container (and you have to think about user ids and permissions)
(2) You need some port mapping for the AvNav port - and optionally for ports of used plugins
(3) SK interworking should work - but will need username+password
(4) You should not install avnav-raspi (makes no sense)
(5) If you would like to access hardware you would need to map this into the container (serial, USB,can)
(6) if you would like to use canboat you need to think about a setup for that one
(7) You need to think about restart handling - either not possible from the UI or with some additional script around
(8) Software update handling. The updater will not work out of the box (currently requires systemd). So either go for command line updates - or recreate the docker container. The latter could have some problems if migration scripts are part of the update (seldom but possible).

tkurki, 2025-08-15:
People using Docker should be already familiar with all of your (very valid) points. Or if they are not they will be if they persist….

wellenvogel, 2025-08-15:
so we are at the same page ;-)