wellenvogel/avnav #271: Computed NMEA data (APB, RMB) sent to Signalk are not correctly parsed

dpazz, 2023-04-27:


wellenvogel, 2023-04-27:
Could you please provide the details of your configuration?
So what base system, which versions of AvNav and Signalk, which connections are configured,...
Looks very strange as NMEA is no Json....

dpazz, 2023-04-28:
Good morning here's the config.
Openplotter install on:
-SBC Rock4SE 4Gb 128Gb microsd (a raspi-like SBC choice due to raspi4c shortage)
-OS arm64 Debian 11 Bullseye headless
-AvNav as of  AvNav updater screenshot:
[image]
-SignalK as of Openplotter settings screenshot
[image]
The aim of the project is to substitute an old monocrome chartplotter with openplotter using AvNav as the navigation interface (via an android tablet) and SignalK as the central data administration interfacing old and new instruments, engine, battery packs (ther'is also a by side sensor project based on ESP32 MCUs connected to signalK). What I would like is to give the tiller Autopilot (a Simrad TP32) all the driving nmea0183 sentences it requires. So far I was able to do this using AvNav directly driving the serial interface connected to the TP32 (now I'm testing at home with a PC receiving from serial). I'm trying to do the same with SignalK driving the the serial but SignalK isn't able to build the APB or RMB sentences so I'trying to get them from AVNav but it seems tat Avnav sent these sentences as "pure" nmea0183 strings and SignalK expects them as "deltas"....

dpazz, 2023-04-28:
Te servers config are here:
-AvNav:
<?xml version="1.0" ?><AVNServer>

<Workers>

<AVNHttpServer httpPort="8080" index="/viewer/avnav_viewer.html" navurl="/viewer/avnav_navi.php">
    <MimeType extension=".avt" type="text/plain"/>
    <MimeType extension=".log" type="text/plain"/>
</AVNHttpServer>

<AVNTrackWriter interval="10" mindistance="50"/>

<AVNSocketWriter blackList="signalk" enabled="True" filter="" maxDevices="5" minTime="50" name="tosignalk" port="28628" read="False" avahiEnabled="False" address="0.0.0.0"/>
<AVNSocketReader enabled="True" filter="" host="localhost" name="signalk" port="10110" writeOut="False"/>
<AVNPluginHandler>
<builtin-Sail_Instrument TWD_filtFreq="0.2"/>
<system-more-nmea sourceName="more_nmea" WMM_FILE="WMM2020.COF" WMM_PERIOD="10" computePeriod="0.5" NewNMEAPeriod="1" FILTER_NMEA_OUT=""/>
</AVNPluginHandler>

<AVNChartHandler/>

<AVNRouter name="AVN-router" computeRMB="True" computeAPB="False"/>

<AVNNmeaLogger/>
<AVNImporter/>

<AVNCommandHandler>
     <Command command="sudo shutdown -P" name="shutdown"/>
     <Command command="/bin/sh $BASEDIR/../raspberry/sound.sh 90%" name="sound" repeat="1"/>
</AVNCommandHandler>

</Workers>
<AVNSignalKHandler decodeData="True" fetchAis="True" fetchCharts="False" migrated="True" receiveNotifications="True" sendData="True" uuid="5991cb59-26e6-402b-be4e-04aeb1e75992" userName="pi" password="raspberry" name="Avnav-Sk-handler"/>

<AVNSerialWriter name="Autopilot" port="/dev/ttyUSB0" stopbits="1" xonxoff="False" combined="False" filter="$XTE,$APB" numerrors="20" autobaudtime="5.0" enabled="False"/>
</AVNServer>

dpazz, 2023-04-28:
<?xml version="1.0" ?><AVNServer>

<Workers>

<AVNHttpServer httpPort="8080" index="/viewer/avnav_viewer.html" navurl="/viewer/avnav_navi.php">
    <MimeType extension=".avt" type="text/plain"/>
    <MimeType extension=".log" type="text/plain"/>
</AVNHttpServer>

<AVNTrackWriter interval="10" mindistance="50"/>

<AVNSocketWriter blackList="signalk" enabled="True" filter="" maxDevices="5" minTime="50" name="tosignalk" port="28628" read="False" avahiEnabled="False" address="0.0.0.0"/>
<AVNSocketReader enabled="True" filter="" host="localhost" name="signalk" port="10110" writeOut="False"/>
<AVNPluginHandler>
<builtin-Sail_Instrument TWD_filtFreq="0.2"/>
<system-more-nmea sourceName="more_nmea" WMM_FILE="WMM2020.COF" WMM_PERIOD="10" computePeriod="0.5" NewNMEAPeriod="1" FILTER_NMEA_OUT=""/>
</AVNPluginHandler>

<AVNChartHandler/>

<AVNRouter name="AVN-router" computeRMB="True" computeAPB="False"/>

<AVNNmeaLogger/>
<AVNImporter/>

<AVNCommandHandler>
     <Command command="sudo shutdown -P" name="shutdown"/>
     <Command command="/bin/sh $BASEDIR/../raspberry/sound.sh 90%" name="sound" repeat="1"/>
</AVNCommandHandler>

</Workers>
<AVNSignalKHandler decodeData="True" fetchAis="True" fetchCharts="False" migrated="True" receiveNotifications="True" sendData="True" uuid="5991cb59-26e6-402b-be4e-04aeb1e75992" userName="pi" password="raspberry" name="Avnav-Sk-handler"/>

<AVNSerialWriter name="Autopilot" port="/dev/ttyUSB0" stopbits="1" xonxoff="False" combined="False" filter="$XTE,$APB" numerrors="20" autobaudtime="5.0" enabled="False"/>
</AVNServer>

dpazz, 2023-04-28:
[avnav_server-20230428.xml.txt](https://github.com/wellenvogel/avnav/files/11351310/avnav_server-20230428.xml.txt)
[settings.json.txt](https://github.com/wellenvogel/avnav/files/11351373/settings.json.txt)
[sk-to-nmea0183.json.txt](https://github.com/wellenvogel/avnav/files/11351382/sk-to-nmea0183.json.txt)

free-x, 2023-04-28:
you defined your "avnav_tosignalk" interface in SignalK as "SignalK"
type. This interface schould be  "NMEA0183"

Greetings
free-x

free-x, 2023-04-28:
btw: you have defined serial interface to autopilot (/dev/ttyUSB0) in
both AvNav and SignalK. This will not work. You have to decide who
plays multiplexer role. AvNav in Openplotter is only "client"  ....

dpazz, 2023-04-28:
> btw: you have defined serial interface to autopilot (/dev/ttyUSB0) in both AvNav and SignalK. This will not work. You have to decide who plays multiplexer role. AvNav in Openplotter is only "client" .... пт, 28 апр. 2023 г. в 10:17, Oleg Roitburd ***@***.***>:
> […](#)

Yes I know, but for now it is disabled in AvNav. As I wrote before if I let AvNav to drive the seria everything works fine. I' trying to do this from SignalK....

dpazz, 2023-04-28:
> I'm wondering about the connections: Did you change them by hand? Normally the installer should have been setting up everything correctly... Which version of the avnav installer do you have (visible in openplotter)?

3.07-stable is the installer release.
As for the connection conf I only added manually the avnav_tosignalk connection just to try a "witer" back connection (I was unexpert of the AvNav/SignalK combi and I didn't know that this was already provided by the std openplotterconf).
Let me give a try deleting this connection or setting it as NMEA0183 as free-x suggests ....

closed 2023-04-28 by wellenvogel