wellenvogel/avnav #273: AIS didn't show (Android)

Izzul24, 2023-05-11:
Hi, I tried AvNav Android and use TCP Reader to receive AIS data. But when I was open the map, it didn't show at all. The decoder said it received AIS data and some targets. I have tried the same source of data on linux and windows it runs fine.

free-x, 2023-05-11:
Hi,

which AIS source did you use? Ist it local? Public? Could you provide dump of AIS Stream? 

```same source of data on linux and windows it runs fine```

Under Linux:
nc <host> <port> | grep VDM

Greeting
free-x

Izzul24, 2023-05-30:
I use version 20230530 running by emulator on Android Studio. I also tried version 20220819.
Here I use the same TCP source on windows and android.
These are my screenshots.
[image]
[image]
[image]

free-x, 2023-05-31:
are you sure that AIS stream is not blocked by firewall?

wellenvogel, 2023-05-31:
Please answet the questiins - esoecialky about the source.
Are you sure that there are targets in the range you defined?

Izzul24, 2023-06-01:
Yeah I'm pretty sure. My source is also a station. I've tested it with set GPS location to a port and the default OSM chart. At the same time I also running AvNav web to the exact location that in my screenshots it showed the AIS just fine. I set the range to max (1000 nM) and still nothing showed up. I'd once tested it with net locator, in the same room as my phone still it didn't showed up.

This is my AIS stream that I've got from android studio:
1,"speed":0.05144444444444445,"course":181,"heading":107,"type":3,"age":54.58}
D/Avnav.AisStore: omitting ais {"mmsi":"352927000","lon":112.66104999999999,"lat":-7.060546666666666,"status":0,"speed":6.8421111111111115,"course":12,"heading":15,"type":3,"age":1.075}
D/Avnav.AisStore: omitting ais {"mmsi":"525900619","lon":112.65356666666666,"lat":-7.115483333333333,"status":5,"speed":0,"course":344,"heading":184,"type":3,"age":68.034}
D/Avnav.AisStore: omitting ais {"mmsi":"525023327","lon":112.70671166666666,"lat":-7.181283333333334,"status":0,"speed":0.05144444444444445,"course":7,"heading":511,"type":1,"age":0.479}
D/Avnav.AisStore: omitting ais {"mmsi":"525001047","lon":112.72982333333333,"lat":-7.207498333333334,"status":3,"speed":0,"course":0,"heading":173,"type":1,"age":10.867,"imo_id":9403970,"callsign":"PLPW   ","shipname":"BIMA SAKTI UTAMA    ","shiptype":90,"destination":"SURABAYA            ","length":60,"beam":12,"draught":3.8}
D/Avnav.AisStore: omitting ais {"mmsi":"525015894","lon":112.72442666666667,"lat":-7.204021666666667,"status":8,"speed":0,"course":172,"heading":511,"type":1,"age":22.966}
D/Avnav.AisStore: omitting ais {"mmsi":"525015659","lon":112.73801166666667,"lat":-7.196755,"speed":0,"course":0,"heading":511,"type":18,"age":43.745}
D/Avnav.AisStore: omitting ais {"mmsi":"525001064","lon":112.714395,"lat":-7.180908333333333,"status":0,"speed":0,"course":0,"heading":511,"type":1,"age":106.864}
D/Avnav.AisStore: omitting ais {"mmsi":"525100143","lon":112.72685499999999,"lat":-7.188675,"status":15,"speed":0.05144444444444445,"course":7,"heading":511,"type":1,"age":10.158}
D/Avnav.AisStore: omitting ais {"mmsi":"538002485","lon":112.688885,"lat":-7.187353333333333,"status":5,"speed":0,"course":354,"heading":126,"type":3,"age":38.242}
D/Avnav.AisStore: omitting ais {"mmsi":"357404000","lon":112.76424,"lat":-6.829876666666666,"status":1,"speed":0.05144444444444445,"course":176,"heading":104,"type":3,"age":116.583}
D/Avnav.AisStore: omitting ais {"mmsi":"777777772","lon":112.67094333333333,"lat":-7.170723333333333,"status":0,"speed":0.05144444444444445,"course":185,"heading":511,"type":1,"age":44.714}
D/Avnav.AisStore: omitting ais {"mmsi":"525600360","lon":112.69853333333334,"lat":-7.179553333333333,"status":1,"speed":0,"course":360,"heading":104,"type":3,"age":88.102}
D/Avnav.AisStore: omitting ais {"mmsi":"525019255","lon":112.73127666666666,"lat":-7.18046,"speed":0.05144444444444445,"course":0,"heading":511,"type":18,"callsign":"YGUE   ","shipname":"","shiptype":70,"length":53,"beam":10,"age":82.293}
D/Avnav.AisStore: omitting ais {"mmsi":"525401965","lon":112.66640000000001,"lat":-7.129616666666666,"status":9,"speed":0,"course":339,"heading":511,"type":1,"imo_id":8594617,"callsign":"YDWW2  ","shipname":"BUNGA TERATAI 168   ","shiptype":70,"destination":"SURABAYA","length":0,"beam":0,"draught":4.1,"age":0.943}
D/Avnav.AisStore: omitting ais {"mmsi":"525020278","lon":112.68664333333332,"lat":-7.170523333333334,"status":0,"speed":0.05144444444444445,"course":115,"heading":511,"type":1,"age":17.251}
D/Avnav.AisStore: omitting ais {"mmsi":"525012233","lon":112.72087666666667,"lat":-7.191063333333333,"speed":0,"course":271,"heading":511,"type":18,"age":47.754,"callsign":"JZIU   ","shipname":"","shiptype":80,"length":83,"beam":8}
D/Avnav.AisStore: omitting ais {"mmsi":"525100815","lon":112.67445833333333,"lat":-7.1546183333333335,"speed":0,"course":0,"heading":511,"type":18,"age":14.712}
D/Avnav.AisStore: omitting ais {"mmsi":"525700374","lon":112.726665,"lat":-7.208586666666666,"speed":0,"course":116,"heading":511,"type":18,"age":53.008}
D/Avnav.AisStore: omitting ais {"mmsi":"525018127","lon":112.71367166666667,"lat":-7.17929,"status":0,"speed":0.15433333333333335,"course":85,"heading":511,"type":1,"age":8.691}
D/Avnav.AisStore: omitting ais {"mmsi":"123456789","lon":112.79504,"lat":-7.2788916666666665,"speed":0.2057777777777778,"course":171,"heading":511,"type":18,"callsign":"","shipname":"LORAIS              ","shiptype":0,"length":0,"beam":0,"age":26.644}
D/Avnav.AisStore: omitting ais {"mmsi":"525101059","lon":112.72971,"lat":-7.2074766666666665,"status":3,"speed":0,"course":240,"heading":511,"type":1,"age":5.391}
D/Avnav.AisStore: omitting ais {"mmsi":"525025049","lon":112.81349666666667,"lat":-6.878276666666666,"status":1,"speed":0.05144444444444445,"course":189,"heading":85,"type":3,"age":119.848}
D/Avnav.AisStore: omitting ais {"mmsi":"525400324","lon":112.73030166666668,"lat":-7.206836666666666,"speed":0,"course":0,"heading":511,"type":18,"age":34.432}
D/Avnav.AisStore: omitting ais {"mmsi":"525025043","lon":112.68571666666666,"lat":-7.003763333333334,"status":0,"speed":5.041555555555556,"course":25,"heading":28,"type":1,"age":24.337}
D/Avnav.AisStore: omitting ais {"mmsi":"525003280","lon":112.671035,"lat":-7.1706183333333335,"status":8,"speed":0,"course":85,"heading":511,"type":1,"age":15.58}
[... cut ...]
D/AvNav:Decoder: getGpsData: {"mode":1,"lat":-7.204116666666667,"lon":112.72513333333333,"course":0,"speed":0,"time":"2023-06-01 03:07:26.000+0000"}
D/Avnav:AndroidPositionHandler: location: changed, acc=5.0, provider=gps, date=Thu Jun 01 10:07:27 GMT+07:00 2023
D/Avnav: Decoder: RMC sentence, valid=true
D/Avnav: Decoder: external position [07.2041167 S, 112.7251333 E, 0.0 m]
D/AvNav:Decoder: Decoder: location: Location[null -7.204117,112.725133 hAcc=??? et=?!? vel=0.0 bear=0.0 vAcc=??? sAcc=??? bAcc=???]
D/Avnav.AisStore: getAisData dist=1852000.0

wellenvogel, 2023-06-01:
Could you please attach the raw NMEA data that you feed into your box?
And for the Debug log: Please attach this as a file. Otherwise it's a bit hard to get all the info from the formatted text as characters are missing.
Maybe you could add some debug output just to see which center locations are used to query AIS data.
In AisStore.java add after line 220 (https://github.com/wellenvogel/avnav/blob/c5cca8bf7324971a07ce8981102fc6b00102a972/android/src/main/java/de/wellenvogel/avnav/worker/AisStore.java#L220):
```
for (Location l: centers) {
            AvnLog.d(LOGPRFX, "center=" +l);
        }
```

Izzul24, 2023-06-05:
Sorry for the Debug Log earlier.  Here are some NMEA data and the Debug Log
[log.txt](https://github.com/wel
[NMEA.txt](https://github.com/wellenvogel/avnav/files/11649482/NMEA.txt)
lenvogel/avnav/files/11649440/log.txt)

wellenvogel, 2023-06-06:
Many thanks. This way it was easy to find the problem.
Strange enough it seems that no one was using AvNav android with AIS data on a southern latitude (i.e. below 0) until now...
I will create a daily a bit later today that will contain a fix.

Izzul24, 2023-06-07:
That's really great! Thanks a lot, now it's run fine.
But sorry, for last question. When I was trying to build the APK, it seems to have error on the AndroidPositionHandlerTest and WorkerTest. It affects a lot of function, I'm not sure what's going on. Here are some of the error::

wellenvogel, 2023-06-07:
For now just forget about the tests...
I tried to build some test env - but several times gave up on it.
I guess in a first step I need to port everything to the newest android libs (androidx).

wellenvogel, 2023-07-03:
release 20230702

closed 2023-07-03 by wellenvogel