wellenvogel/avnav #275: Installation on Bookworm failed

free-x, 2023-05-31:
Hi @wellenvogel ,

i tried to install avnav deb package on Debian/Bookworm. And it's failed
```
apt install avnav
....
creating basedir /var/lib/avnav
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation 
or OS distribution provider. You can override this, at the risk of breaking your
 Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
dpkg: error processing package avnav (--configure):
 installed avnav package post-installation script subprocess returned error exit
 status 1
```

Regards
free-x

free-x, 2023-05-31:
I think a solutions could be to drop usage of pip and include python3-bluez as "Required"

wellenvogel, 2023-05-31:
This did not work on OpenSuse...

free-x, 2023-05-31:
what's about
```
if [ ! -x /usr/bin/dpkg ]
then
   /usr/bin/python3 -m pip install pybluez
fi
```
then I install "python3-bluez" in avnav-stage

wellenvogel, 2023-07-03:
release 20230614

closed 2023-07-03 by wellenvogel