wellenvogel/avnav #278: Request: Add the ability to edit the anchor alarm parameters

OliverWieland, 2023-06-10:
Sometimes it is useful to change the alarm radius afterwards. To do this, at the moment, I have to
- write down the current parameters (alarm distance, current distance, bearing)
- stop the anchor alarm
- start the anchor alarm again and set the saved parameters (with the required modifications).

It would be very nice, if there would be the ability to edit the alarm parameters without stopping and restarting the anchor alarm.

wellenvogel, 2023-06-10:
You can do it more easily this way:
1. On the map page bring up the waypoint buttons ( click on lower left widgets)
2. Center to waypoint with those buttons (make the position of the anchor the map center)
3. Stop anchor watch with this buttons
4. Bring up waypoint buttons again
5. Start anchor watch with new radius using "center" leaving distance and bearing at 0 (your map center is where the anchor is...)
Still not perfect but without any need to write something down.
But we can keep this issue open, I agree that it would make sense to have an edit option.

wellenvogel, 2024-04-23:
continued/implemented with #303.

closed 2024-04-23 by wellenvogel