wellenvogel/avnav #281: AVNav not finding OpenCPN charts

VibroAxe, 2023-06-21:
I have a series of o-charts installed in OpenCPN however I cannot convince AvNAV to connect and use them. As far as I can tell this is the only way to use O-charts now as you can no longer download the zip.
[image]

Any suggestions would be appreciated (dongle license). Happy to provide debug but not sure where best to find it?
[image]

free-x, 2023-06-21:
hi,

1. it's always nice when you include a log file. You can find provider.log* in $HOME/avnav/data/ocharts/
2.  Am I right that you have downloaded and installed o-charts sets via OCPN plugin?
3. Did you try downloading ZIP with charts, unzip and add in OCPN manually (Settings->Charts->Charts Files->Add Directory)

Regards
free-x

VibroAxe, 2023-06-22:
Thanks @free-x useful start point

provider.log is here> https://gist.github.com/VibroAxe/a74d2fdba86094aca9268133417c587c

Potentially relevant line is line 20
```
2023/06/22-10:28:46.959-0x7fa12197c0-INFO-found chart dir /home/user/charts/oeuSENC-UK in ocharts but not on opencpn level```

The oeuSENC-UK chart is the one I am expecting to see, which is loaded into openCPN and is loaded correctly in openCPN?

wellenvogel, 2023-06-24:
You are completely right....
There is some bug that I was not able to fix until now.
But as far as I know you still should be able to require the charts on the o-charts page to get a zip download link.
Could you double check please?
Otherwise I guess we need to get in contact with o-charts.

Maybe you can try an intermediate solution on the command line:
(1) create a folder oeuSENC-UK at /home/pi/avnav/data/ocharts/charts
(2) copy all files from /home/user/charts/oeuSENC-UK there
(3) restart the o-charts provider
You could also create a symlink instead of (1) + (2)
```
cd /home/pi/avnav/data/ocharts/charts
ln -s /home/user/charts/oeuSENC-UK oeuSENC-UK
```

wellenvogel, 2023-06-25:
Small addition:
Could you please attach your OpenCPN config file?
Just remove any keys/passwords...

VibroAxe, 2023-06-29:
So the intermediate solution has "sort of" worked. If I create the symlink and restart the chart provider I get the UK chart showing up (woo). However, on avnav restart / sometimes during operation the provider errors out
[image]

provider.log>https://gist.github.com/VibroAxe/a74d2fdba86094aca9268133417c587c#file-provider_crashing-log

opencpn.conf>https://gist.github.com/VibroAxe/a74d2fdba86094aca9268133417c587c#file-opencpn-conf

wellenvogel, 2023-06-30:
There is a new daily out.
This should address both issues.
I.e. the OpenCPN charts should be used even without copy and there should be no errors any more.

VibroAxe, 2023-06-30:
> There is a new daily out.
> This should address both issues.

Fantastic, however I'm not currently seeing an update in avnav updater (BBN-OS). Not sure if this is a sync issue with BBN-OS somewhere or if I need to do something particular to install the daily?

VibroAxe, 2023-06-30:
> > There is a new daily out.
> > This should address both issues.
> 
> 
> Fantastic, however I'm not currently seeing an update in avnav updater (BBN-OS). Not sure if this is a sync issue with BBN-OS somewhere or if I need to do something particular to install the daily?

Scratch that, found the dpkg file and installed. Chart is correctly being found from the opencpn charts folder, currently rebuilding cache so will report back in the morning if it's stable!

wellenvogel, 2023-07-03:
Should be solved with avnav-ocharts-plugin 20230702.

closed 2023-07-03 by wellenvogel