wellenvogel/avnav #282: autohide 2 column buttons on dashboard in landscape mode not oke

PietKnot, 2023-06-22:
i noted that -at least sometimes- the autohide bottons on the dashboard page, works partly: the buttons are hidden, and a white area appears not filled by an extented display-area for the widgets:

 The non-hidden buttons on a screen 1080 x 540 px
[image]

en now in hidden mode:
[image]

The option "show shade when button hidden is not marked. In one colunm mode (MOB-button not present when not navigating at home, all is normal.

wellenvogel, 2023-06-24:
Unfortunately I cannot reproduce the problem here.
Which device and which browser are you using?

wellenvogel, 2023-06-24:
OK, problem reproduced:
Only with 2 button columns.

PietKnot, 2023-06-29:
works correct now; thanks

wellenvogel, 2023-07-03:
release 20230702

closed 2023-07-03 by wellenvogel