wellenvogel/avnav #283: handling too many buttons for two button rows

PietKnot, 2023-06-22:
reporting now a very small issue: when too many buttons are to be displayed in 2 widget rows in landscape mode, it would be nice to have a third row:

[image]

There are actually 11 buttons displayed in portrait mode, but in landscape mode only 10 are visible and the Back button missing. I know that i could reduce the button size but they are actually already small (on 4 inch screen), thus maybe it's easy to implement, but don't bother too much about it.

wellenvogel, 2023-06-24:
OK, also reproduced...
The button auto resizing does not work any more.
In this situation the buttons should be reduced in size.

PietKnot, 2023-06-29:
works almost: the buttons are resized (now 6 in one columN), but the MOB-button is "lost", e.g. not displayed anymore; it was the tenth button of the 11  in portrait mode; There is space left is the second column

wellenvogel, 2023-06-29:
That would be strange. Are you sure that you have a valid position? Otherwise no MOB button at all...

PietKnot, 2023-06-29:
Yes; rather sure. Just made the verification at home again with internal GPS on Android phone.

Portrait mode: 11 buttons including MOB and active position
[image]

Lanscape mode: MOB missing; made at the same time after the above one.
[image]

free-x, 2023-06-30:
Hi,

i can't confirm this behavior with my FHD/FHD+ gadgets.
Which resolution of screen do you have?

Regardings
free-x

wellenvogel, 2023-06-30:
I cannot reproduce this on my side too.
Which device are you using? Looks like a rather old android device. 
Do you use the native android app? BonjourBrowser? Browser?

Any way the "reduce size" handling is more like a "last resort" if you accidently set the button size too big.
So if this hits only some rather old device I would not invest too much on it. Maybe the browser does not trigger some resize - or it reports wrong dimensions...

Could you cross check if the button will become visible if you leave the page and go back?

wellenvogel, 2023-07-02:
Ok, inverstigated a bit further:
I was able to reproduce some similar situations and made an additional correction.
A daily build is on the way - maybe you could retest using the one from today.

PietKnot, 2023-07-02:
i will do further tests bij reducing the button-size to see if the MOB-button was possibly the 7th button on the right column with 6 buttons.

But first the requested info on the Android. it's not a rather old Android device (actually a  Cubot mini KingKong 2) running Android 10 with graphical spec:

4.00 inch, 18:9, 450 cd/m2
1080 x 540 pixel 302 PPI,
kapazitiver Touchscreen, IPS,
glimmend: ja, 60 Hz

wellenvogel, 2023-07-03:
release 20230702

closed 2023-07-03 by wellenvogel

PietKnot, 2023-07-03:
Excellent, well done; all buttons are visible now with 7 buttons in one column (although 6 buttons would work as well see above). Thanks for the "last resort" functionality that allows for larger buttons in portrait mode and smaller in landscape mode.

P.s. a button-size settting for landscape-mode would have done it as well and maybe easier to implement.