wellenvogel/avnav #284: Location of Version information

PietKnot, 2023-06-24:
to find the version information of the running AvNav on Android, you have "search"  and will find it on:

MainPage / Server Status / Android / "small "i" icon.
Here the version info is present together with Info on Licence and Privacy.

One level higher thus
MainPage / Server Status / "i" button
Here is the same info on License and Privacy is also present, but without the version info.

I would suggest to add the version info at the higher level also (and maybe remove the info on the Android settings ).

wellenvogel, 2023-06-24:
Valid point. I will have a look.

PietKnot, 2023-06-29:
Works and thanks; would still remove the duplicated info on the lower level, as it is exactly the same (exact minor layout difference)

wellenvogel, 2023-06-29:
Maybe later on - but it's good practice to have it there (just consider initial start...).

closed 2023-06-29 by wellenvogel