wellenvogel/avnav #285: No upload capability

antonical, 2023-06-25:
I cannot see anyway to upload a file on the file page. There  is no upload icon on the menu?

How can I import routes etc.

Cheers
Tony

wellenvogel, 2023-06-25:
Could you please be a bit more verbose?
What page do you mean (there is no "file page")?
Which Version (linux/windows/android)?
Are you sure you are in "connected mode" (button with the plug symbol on the main page green)?

antonical, 2023-06-25:
Andreas,

It is a server install on RPI4,

Connected to it via chrome browser on Ubuntu 20.04 

I am not at the boat at present but will try the plug symbol tomorrow.

But on yhe files/download page there is no upload icon.

Will revert tomorrow.

Thank you for your prompt response.

antonical, 2023-06-29:
Andreas,

Ok I had missed the plug icon and being 'online' on both the nav computer browser session and the cockpit tablet that seems to have solved the issue and we can now upload routes etc. 

OAN,  are there any plans to incorporate grib overlays onto the live active route. 

Cheers

wellenvogel, 2023-06-29:
Hmm - grib files:
Would need some js library for doing that. It's on the wish list somehow..
Maybe you would like to open an issue for that.
Closing this one as solved.

closed 2023-06-29 by wellenvogel