wellenvogel/avnav #289: Hover text on Icons would be useful

antonical, 2023-07-06:
This is an enhancement request to put hover text on the icons around the application.

Cheers

marcelrv, 2023-08-20:
This would be so helpful. I like the responsiveness of avnav, but even though I use it quite frequently I am still completely oblivious on the icons and what they do. It seems I always click a few and go back before I found the right one.

wellenvogel, 2023-08-20:
The problem is:
It will not work on touch devices.
And this is the main focus.
So for me there is a trade off between effort an positive effects...

marcelrv, 2023-08-20:
Does it have negative efforts for touch devices?
I am willing to try to contribute to the code of you can give a hint on how it is done.

wellenvogel, 2023-08-22:
I guess the code itself is not that much of a problem.
The bigger part is to collect all the button names (doc/browser dev tools) and create a (meaningful) text for each of them.

quantenschaum, 2024-02-08:
Yes, it's work, but it would make it much easier to get to know the software if there were some hints when hovering the buttons. People will often test it on a device with a mouse attached, I think. And you can attach a mouse to a tablet as well.

There is software that doesn't need a manual because the gui is so intuitive. AvNav is great but at first not really intuitive. Took me a while find out what certain buttons do. You can and should of course RTFM but a little help text here and there would speed things up a lot. The help text behind the little (?) in the config is great!

wellenvogel, 2024-02-08:
See my comments above.
Basically I agree.
If somebody is willing to take the job of writing meaningful help texts for all buttons...
The implementation itself is rather simple.
And still my major focus is on touch devices.

antonical, 2024-07-21:
I am happy to write the help texts for each button. Can you let me know how you want then sent to you?

Shall I screenshot and mockup how it might look or just a file with the Icon and screen text.

wellenvogel, 2024-07-22:
Basically it will require some json or yanl files with the button names (like in the doc) as keys and the description as values.
But I'm still looking for some integration with an internationalization framework...
Anyway a json/yanl with the english descriptions would be a good starting point.

nmeyne, 2024-08-03:
Our (interim) club solution to this is to make up a laminate card with the AvNav  icons and text on it, plus some other tips.  It's low tech, but works well in the wet! 😉

wellenvogel, 2026-11-02:
implemented with [20261102](https://www.wellenvogel.de/software/avnav/newDoc/latest/en/installation/versions/20261102.html)

closed 2026-11-02 by wellenvogel