wellenvogel/avnav #293: Have had many issues updating charts

antonical, 2023-07-12:
Recently I decided to update my charts using the o-charts plugin. 02072023

When I started this process I was running an older version of avnav.

Updating charts was unsuccessful.

Eventually, I gave up and removed all charts from the ~/avnav/data/ocharts/charts directory and all cache files.

I then upgraded the avnav base system and all plugins and started fresh.

I still had issues where the chart would upload (100% on the progress bar) then start the validation/verifiy process and eventually ask to restart. Some times the chart would upload and then I would get a pop up saying server timeout during the validation/verification process. This is weird as the pi is hardwired to a network switch as is the client machine.

HWn I checked if the chart had indeed uploaded the chart it was not found in the chart directory and no additional space used. So no idea where it went. Nothing to show it ever happened on screen.

It took multiple loops around this process to eventually load all my charts with the charts seemingly randomly failing the succeeding on the next attempt.

I was monitoring the pi via another terminal and it had plenty of cpu and free memory.

Any suggestions as to what is happening?

Cheers
Tony

wellenvogel, 2023-07-13:
Basically your issue is at the wrong repo.
It's an ocharts plugin issue...
Anyway I try to answer:
(1) would be good to know what you updated and what went wrong.
(2) Which image/hardware do you run?
(3) Which charts did you try?
(4) From what you described I would guess that it could be a slow SD card. They tend to become slower after using them for longer times. If you run a top command during the upload it would probably show high waitio (wa).

If the upload/check is failing, the charts are removed to avoid dead files.
What you could try:
Unpack the zip directly in ~/avnav/data/ocharts/charts.
This should create a new subfolder.
Afterwards restart to reload the charts.
Have a look at ~/avnav/data/ocharts/provider.log for errors.

wellenvogel, 2023-07-13:
(1) - updating the whole system is always a chance for some issues (but not related to your problems) - nothing I would really recommend as we cannot test all the possible combinations(better use a newer image). For the AvNav stuff - just use the updater.
(2) Server Image means AvNav image - right?
Basically you can just copy the zip file to ~/avnav/data/ocharts/charts any way you want...
And then unpack it there (maybe just have a look at top - waitio...). Would be interesting to get the runtime of the unzip process.
Do the restart afterwards.
The only drawback is that the restart would be slowed down if you e.g. have the wrong key for the charts - this is normally checked immediately after upload.
Basically the process that runs after the upload reached 100% is the one with most IO (SD card) activity - int unpacks all the chart files from the zip.
Just keep the chart zip files somewhere (laptop?, usb-stick,...) this will also be a back up for you as you cannot download them any more if your license is expired - but you can still use them (and - if necessary- reinstall them).
If you still encounter problems - the provider.log would be helpful.

antonical, 2023-07-14:
Thanks Andreas,

1) Ok got it. We usually use the avnav updater screen anyway. But this pi had not been updated for a while.
2) Yes I believe it came from the Current Image link on your page and was around 1GB
Ok I get it now with the uploader.
Good advice on the backup of the original downloads of the chart zips.

Will report any future issues on the plugin page,

Cheers

wellenvogel, 2024-02-19:
No further activity - so closing this.

closed 2024-02-19 by wellenvogel