wellenvogel/avnav #295: Reverse Proxy Support (feature request)

rahra, 2023-07-13:
I try to run Avnav behind a reverse proxy. I'd like to proxy all requests going to `http://myserver/avnav/.*` to `http://avnavserver:8080/$1`.

It does not work because several URIs seem to be embedded hard-coded. So it expects to be run on the root URI `/` and not somewhere below. This also applies to the plugins, e.g. the O-Charts plugin which is found on port 8082. But it also expects to be run on the root URI.

It would be nice to have configurable paths for all that.

rahra, 2023-07-14:
Yes course this is not of high priority.
To accomplish this it would be necessary to make the URI which is sent back to the client configurable.

Here is my experimental config. It assumes that the Apache working as reverse proxy is running on the same host as Avnav. If not, %HTTP_HOST would have to be modified.
Furthermore this doesn't make the whole thing work as a real reverse proxy but only partially because at least the connection to the O-Charts plugin still goes directly to port 8082. I could not yet find a good way to deal with this (except by excessive on-the-fly regex substitutes of the responses).
With this config Avnav is available under http://_reverseproxyname_/avnav/viewer/avnav_viewer.html
However, even for me this has no priority, although I brought up this idea ;)

```Apache
        ##### AVNAV REVERSE PROXY #####
        RewriteEngine on
#       LogLevel alert rewrite:trace6
        RewriteRule "^/avnav/(.*)$" "http://%{HTTP_HOST}:8080/$1" [P,L]
        RewriteRule "^/viewer/(.*)$" "http://%{HTTP_HOST}:8080/viewer/$1" [P,L]
        RewriteRule "^/plugins/(.*)$" "http://%{HTTP_HOST}:8080/plugins/$1" [P,L]
        RewriteRule "^/user/viewer/(.*)$" "http://%{HTTP_HOST}:8080/user/viewer/$1" [P,L]
        RewriteRule "^/remotechannels/0$" "ws://%{HTTP_HOST}:8080/remotechannels/0" [P,L]
        ProxyPassReverse "/avnav/" "http://%{HTTP_HOST}:8080/"

        RewriteRule "^/avnav-ocharts/(.*)$" "http://%{HTTP_HOST}:8082/$1" [P,L]
        RewriteRule "^/status/(.*)$" "http://%{HTTP_HOST}:8082/status/$1" [P,L]
        ProxyPassReverse "/avnav-ocharts/" "http://%{HTTP_HOST}:8082/"
        ##### /AVNAV REVERSE PROXY #####
```

**Background**
I run a bunch of other services on the same machine (which is my boat computer) such as SignalK, Grafana, some PHP-based tools under Apache and some others. Several of theses services come with their own standalone HTTP server on some TCP port. So my idea is to hide all that behind the Apache to have one central instance through which I can access all of that.
This also would make sense in respect to security, and I could even put a TLS certificate in front of all that.

wellenvogel, 2023-07-14:
Maybe it could also make sense to use the x-forwarded... headers in AvNav to handle such situations.

rahra, 2023-07-14:
Yes that could be an option to deal with it.
Although actually to put it on a separate VHost under the root URI would work right now. But using the X-Forwarded from the reverse proxy together with the URI to be configurable would make the config straight forward.