wellenvogel/avnav #296: Grib file overlay

antonical, 2023-07-14:
Hello Andreas,

Is there a plan to allow grib files to be integrated or overlaid in avnav. For instance files from xygrib.

Currently we run xygrib on the client nav computer and display avnav in a browser.

Cheers

wellenvogel, 2023-07-14:
There already have been questions on that.
It would be relatively easy if there would be a java script library to show the grib files.
But until now I did not find any.
Another option would be to render them on the server - but this is much more effort.

Fietsenwiel, 2023-07-20:
Does this help: https://github.com/spidru/JGribX ? 
**JGribX was originally designed to be a Java library**, meaning that it did not have any useful functionality when run as a standalone app. However, a command-line interface is currently being developed which allows JGribX to be used as a standalone app via command-line.

wellenvogel, 2023-07-20:
Not really a lot.
It's just java - not java script.

Fietsenwiel, 2023-07-20:
Ah, Ok.

I already thought it wouldn't be so easy :)

Thanks!

antonical, 2024-07-21:
Is it possible to use the opencpn grib plugin as base for this? I am not a coder so not sure.

Right now we use Xygrib and run that on the nav computer. But really trying to consolidate devices as much as possible. As the RPI4 AvNav server is headless not sure how to get gribs on the server side. I guess it would be something core to the avnav server.

wellenvogel, 2024-07-22:
Basically there would be 2 options:
(1) Render the gribs on the server an send the rendered data as a picture to the browser
(2) Extract gribs on the server and add some js based overlay. This would require some js based code for the rendering.

I would prefer (2) - but did not find a promising js lib.

The OpenCPN plugin would be more (1) - but its a huge effort to integrate this (did something similar for o-charts) - nothing I will do again.

Fietsenwiel, 2024-10-13:
Hi Andreas,

I hope I may ask you a question by mail.

I have updated avnav on my raspberry pi to the latest version from the
update function in AvNav.

Now I don't have o- charts anymore.

In AvNav the fingerprint of the ocharts dongle is shown correctly.

Do you know what went wrong?

Thanks in advance.

Lex