wellenvogel/avnav #301: Layout .json files are not deleted.

teesix, 2023-08-04:
I created a few new layouts by entering a new name for the new Layout at the start of the Layout editor (for instance “another layout” and “a third layout”. At the download section on the Layouts page these instances can be seen and can be downloaded. If not current also a trash can is displayed, and there is another delete option after the line representing the instance is pressed. When pressing delete a confirmation popup is shown with cancel and ok buttons. But pressing ok does not result in a delete, the files are still there. Also refreshing or restarting the app doesn’t change it. This was tested on a Samsung Galaxy Tab A (Android 8.1 ) and an Aldocube iPlay40 pro (Android 11), both without SD card, running avnav-20230705, without any other divice connected.

wellenvogel, 2023-08-05:
Thanks for reporting, will have a look.

wellenvogel, 2023-08-05:
Yes, can reproduce it. Not only android.

wellenvogel, 2023-08-06:
There is a new daily at https://www.wellenvogel.net/software/avnav/downloads/daily/20230805/.
Maybe you could try it - it should fix this bug.

teesix, 2023-08-07:
Installed the daily APK on a galaxy Tab A, Android 8.1 . Things work as expected. Bug fixed.

wellenvogel, 2024-05-20:
20240520

closed 2024-05-20 by wellenvogel