wellenvogel/avnav #302: Avnav/O-Charts collides with OpenCPN

rahra, 2023-08-05:
I have the suspicion that Avnav using the O-Charts plugin collides with OpenCPN if both run on the same system.

Observation:
I start Avnav at system startup up with the systemd service file. After the system is booted I start OpenCPN. It lags and eats up CPU and finally complains that something is wrong with the O-Charts (license, system changes, whatever).
As soon as I stop the Avnav service OpenCPN immediately continues to work fine.
After OpenCPN runs I can start the Avnav service again.
Finally both seem to work in parallel but I'm not 100% sure since some lags occure in both programs with time.

Is it possible that the O-Charts servers of Avnav and OpenCPN try to use the same ports or whatever?
I don't know how the communication between the software and O-Charts server works that I could further investigate that.

wellenvogel, 2023-08-05:
Currently there is not much I can do about this issue.
The decoder for o-charts uses a (fixed) pipe at /tmp - and running 2 decoders at the same time could lead to problems.
Basically from my tests it seems to work in most cases if you run both in parallel - but there could be problems as you see.
As I do not control the code that is dealing with this part there is no chance...
I'm working on a rewrite of the o-charts part - and this will also solve the parallel work problem. But this still takes time.

rahra, 2023-08-06:
Ok, good to know.
I'll add `PrivateTmp=yes` to the systemd service file and report my result after testing.

wellenvogel, 2023-08-06:
yes, this seems to be a nice idea.
If it works we can add this to the AvNav openplotter installer.

rahra, 2023-08-06:
Yes it works. I just tested it and it solved the problem that I reported.

wellenvogel, 2023-08-07:
Perfect. Thanks.

rahra, 2023-10-25:
I have the suspicion that Avnav (or the AvnavOchartsProvider) may hang or misbehave if charts are updated within OpenCPN while Avnav is running. I know that this is somewhat vague. Since I discovered this first, I usually stop and restart the Avnav service when I update charts within OpenCPN.

But I can observe this more closely the next time when I update charts.
Is there something specific where I could/should look at?

wellenvogel, 2023-10-25:
For sure this is not going to work without a restart of the ocharts provider (typically no need to restart AvNav).
Just trigger the restart on the charts tab of the provider GUI.

wellenvogel, 2024-05-20:
closed due to no further activity

closed 2024-05-20 by wellenvogel