wellenvogel/avnav #303: Request: Insert a warning dialog before deactivating the anchor alarm

OliverWieland, 2023-08-18:
More than once I had the problem, that I'm preparing the route for the next day while the anchor alarm is activated and by mistake I activate the route. On this time the anchor alarm is deactivated without any information. That can lead to very big problems. Another problem is, that, when there is a lot of swell, the anchor button could be clicked by accident instead a button beneath it by accident.
Therefore I request a warning dialog before deactivating the anchor alarm that gives the possibility to cancel the action. This dialog should be displayed everytime when the anchor alarm is about to be deactivated , no matter in which way this action was initiated.

wellenvogel, 2023-08-21:
Very valid point.
I will think about a solution.

OliverWieland, 2023-08-22:
(1) I'm not sure if it makes sense to show the anchor watch button in the nav page. At least I can't imagine a situation in which I could need it there. And every useless button is one button too much... ;-)

(2) That sounds very good, especially with the possibility to edit the anchor watch parameters

(3) That's exactly what I would expect. Either a complete prevention or at least a dialog to confim the action. But not only in the route editor but on all pages and dialogs where a route can be activated.

(4)I can edit a route when not connected to the server? Didn't know that... :-|

Additionaly I think, the anchor watch is a security feature, which needs a clearer, more eye-catching visualisation when activated. For example an anchor symbol somewhere (sorry, have no idea where and how...)

wellenvogel, 2024-05-20:
20240520

closed 2024-05-20 by wellenvogel