wellenvogel/avnav #304: Browser window with avnav suddenly closes on android

OliverWieland, 2023-08-18:
I use AvNav on a Samsung Tab A6 as routing software when sailing. While sailing and the map (O-Charts, both vector and rasterized) is displayed, sometimes suddenly the browser (Chrome) closes. This seems to happen only, when the map is displayed, at minimum once per day, mostly two or three times per day.
It doesn't happen, when the dashboard is displayed (the dashboard page with the anchor alarm is opened the hole night when I'm anchoring, and there the issue didn't happen for now 4 months).
When I re-open the browser with avnav, all is working again without any issues.

wellenvogel, 2023-08-22:
This is really hard to analyze. Never had something similar. At the end the java script code should not be able to crash the browser at all.
Could you try another browser?
Maybe just go for my [BonjourBrowser](https://play.google.com/store/apps/details?id=de.wellenvogel.bonjourbrowser&hl=gsw&gl=US) and enable it's internal browser (gives you some nice additional features anyway).

OliverWieland, 2023-08-22:
Will try that...

wellenvogel, 2024-02-19:
No further activity - so closing this.

closed 2024-02-19 by wellenvogel