wellenvogel/avnav #305: Browsing through the AIS list is not possible due to refresh

OliverWieland, 2023-08-18:
The AIS list is refreshed every time a new AIS-Sentence is received via NMEA. When I try to browse the AIS list page, for example to look for a specific ship, the constant updating is very annoying. Dependent on the newly received  information the ship order in the list changes more or less completely. 
Therefore I suggest a check box on the AIS list page to deactivate the list update. Additionally a refresh button for a one-time-update would be helpful.
Another enhancement for the AIS list page would be a possibility to save the order type (MMSI, Name, size, type,distance ...)

wellenvogel, 2023-08-22:
This is a good point.
Most probably the easiest solution is to have a "lock" button on the AIS (list) page (with keeping it's state).
So when lock is on - no updates. To trigger an update you would basically double tap the lock button (first tap: unlock + update, second tap: lock again).
What do you think?
Maybe additionally we need an indication if a new AIS warning is computed...

OliverWieland, 2023-08-22:
Yes, that sounds good. Also the new-message-indication would be a nice feature.

wellenvogel, 2023-08-26:
Testing with daily https://www.wellenvogel.net/software/avnav/downloads/daily/20230822/

wellenvogel, 2024-05-20:
20240520

closed 2024-05-20 by wellenvogel