wellenvogel/avnav #309: Ignore unreasonable depth values from SignalK

rahra, 2023-09-02:
In my setup AvNav gets its data from the SignalK server. If the depth transducer is unable to sound any depth (because it is too deep) then SignalK sends `environment.depth.belowTransducer = 42949672.92` [m] which is directly displayed within AvNav (this is close to 2^32). The values `environment.depth.belowKeel` and `environment.depth.belowSurface` are calculated and show similar values.

I think all values which are greater then 12000 can safely be ignored (although there is probably no transducer which could even sound such a depth. The limit could probably be smaller).

wellenvogel, 2023-09-02:
Basically I agree.
On the other hand it would make more sense to address the problem at it's root cause.
Did you already check if the transducer itself sends this value?
Or is there maybe a bug in SignalK?

rahra, 2023-09-02:
No I didn't. I'll first have a look at the values of the transducer. Then I can figure out where the origin of the problem is.

wellenvogel, 2024-04-23:
will close this due to missing responses

closed 2024-04-23 by wellenvogel