wellenvogel/avnav #310: useRhumbline setting doesn't seem to change AP sentences

MichaKersloot, 2023-09-04:
It looks like this setting doesn't change anything in pgn 129284? 

[image]

[image]

[image]

[image]

MichaKersloot, 2023-09-04:
But what does AvNav send that SignalK-N2k does use to create the PGN ? If I look directly in SignalK there also is only the Rhumbline navigation information and not the GreatCircle despite the setting. But getting a screenshot of that was more difficult for me because of all the changes on the screen in the SignalK Data browser.

wellenvogel, 2023-09-04:
It deprnds on how you connect AvNav to SignalK.
So please describe your set up a bit.
If you use only NMEA @free-x '  answer is correct.
If you use the AvNav Signalk handler with write enabled you should see the correct values in SignalK.

MichaKersloot, 2023-09-04:
I use openplotter. I have enabled the SignalK write, but I don't know if the nmea is still enabled. As I'm not at the boat at this moment, could you please tell me what information I should gather next time I'm there?

wellenvogel, 2023-09-05:
I investigated on this issue a bit...
Unfortunately there are several issues.
(1) The signalk to nmea2000 plugin always sends 129284 using the Rhumbline routing parameters. It never reads/sends out the greatCircle values. So you will never see 129284 with type great circle. Maybe this should be addressed as an issue at their [repo](https://github.com/SignalK/signalk-to-nmea2000).
(2) SignalK always interprets the routing data as Rhumbline when decoding NMEA0183 RMB. So whenever RMB is received you will see navigation.Rhumbline... in the data browser
(3) AvNav sends incorrect Names with the SignalK handler towards SignalK for the routing data. There is a typo - so it sends RhumLine instead of Rhumbline.

Now it depends on the connection between AvNav and SignalK what happens:
(a) Only SignalKHandler, no NMEA connection: SignalK will never send out 129284 regardless of the routing mode you use in AvNav (caused by (3) - otherwise it would send if you use routing mode Rhumbline).
(b) NMEA connection from AvNav to SignalK (SocketWriter in AvNav) that at least allows $RMB to pass:
SignalK will always send 129284 with type Rhumbline - caused by (2).

So what is the conclusion:
(A) Bad news: Currently there is no chance to get 129284 with type greatCircle (1).
(B) Once the bug in AvNav (3) is corrected it would most probably make sense to only let the SignalK handler provide the routing data to SignalK and disable the NMEA connection if you want a clear picture.
(C) On the other hand for most use cases it should not really matter which course you use - so letting the NMEA0183 RMB go to SignalK will at least always give you 129284 regardless of the AvNav routing mode.

If you need some more help for the configuration - please attach your avnav_server.xml.

wellenvogel, 2023-09-05:
Ok, I made the necessary corrections in AvNav, a daily is just building. Maybe you would like to try this out.
For SignalK I created a PR: https://github.com/SignalK/signalk-to-nmea2000/pull/79.
If you want to already try this in advance, you would need to replace the file
/home/pi/.signalk/node_modules/signalk-to-nmea2000/conversions/navigationdata.js with the changed one at https://github.com/wellenvogel/signalk-to-nmea2000/blob/master/conversions/navigationdata.js.
Afterwards restart the SignalK server and enable the 129284 output also for greatCircle.

MichaKersloot, 2023-09-12:
Hi,

I've applied you changes and disabled the NMEA0183 connection to reduce clutter. I've found some other issues in the signalk-to-nmea2000 converter, but will continue on that in their repository. Thank you very much for your support!

closed 2023-09-12 by MichaKersloot